sebanc/brunch #1427: Kernels above 5.4 do not work on surface book 1

pheyx, 2022-02-09:
Hi，thanks for the great work.
I installed lastest brunch and chromeos 97 on my surface book 1. everything works except the cameras if booting whith kernel 5.4.
kernel 5.10 or 5.15 would cause boot loops after brunch showing it's rebuilding rootfs. 
I want to try kernels above 5.4 because I heard the cameras may work using kernels after 5.1x.
thanks.

Steso871, 2022-02-12:
Hello!
I have the same problem on my Surface pro 5 (2017) i5.

Is there anything i can do to help get it to work?

sebanc, 2022-02-13:
It should be fixed in the below build:
https://github.com/sebanc/brunch-unstable/releases/tag/r98-unstable-20220213

Steso871, 2022-02-13:
Can confirm both 5.10 and 5.15 is working with my Surface pro 5 (2017) i5 !
Thank you so much for fixing my problem very fast

sebanc, 2022-02-13:
No pb, happy to see that it is fixed :)

Frankygishere, 2022-11-28:
All working (except camera) on Surface Book 1 with 5.15. Is there anyone working on getting the cameras to work?