sebanc/brunch #1433: On my laptop OpenGL v3 - so 99 percent of games does not start

sanek184, 2022-02-11:
Hello. It is a translation.
I have a laptop with an I3 3120M processor
When I install Chromeos - then AIDA64 shows OpenGL V3 - and almost all games do not start. Tried ramus v97 and nautilus v97
In Androids x86 OpenGL V4.2 by default - and all games work
How can I change the version of OpenGL in Chromeos on 4.2?

ksoham622, 2022-02-15:
When you see the login screen on your Chrome OS wait for 2-3 minutes and then enter you password and login.This might work
because android container loads slowly on slow HDD

ksoham622, 2022-02-15:
And use rammus only which is best supported for your processor

sanek184, 2022-02-15:
> And use rammus only which is best supported for your processor

[image]

closed 2026-01-28 by github-actions[bot]