sebanc/brunch #1434: android_init_fix seems to lose effect on the latest r97

zhufucdev, 2022-02-12:
Android container fails to load, and apps won't start without executing chrome://restart
I'm running zork image on Lenovo Yoga 7 Slim, with a ryzen 7 4800u
Here's the log, where line 1669 indicates the load process starting and line 2321 stopping
[messages.txt](https://github.com/sebanc/brunch/files/8053891/messages.txt)
I would be grateful if you could lend me a helping hand

zhufucdev, 2022-02-12:
forgot to note that brunch of r94 seems to work fine, even running r97

zhufucdev, 2022-02-13:
> Have you tried disabling "android_inix_fix" ? You should not need it in theory, it mostly targets old computers with slow hdd.

yes, it's the same
I also suffer from long wifi initialization time with the option disabled, which is not expected on a nvme ssd

sebanc, 2022-02-13:
Well, I am sorry but I see nothing wrong in the log. This wifi driver is supposed to be well supported and this laptop is nearly identical to mine in terms of specs.

The only thing that would make sense to me right now would be a vpn issue.

sebanc, 2022-02-13:
I would say so, the log seems ok, other users with this wifi driver have no issues as far as I know and my own laptop as very similar specs (aside from the wifi driver) so the network environment is my only guess.

I have already seen several issues with the android subsystem which were related to the use of a VPN and iirc some users were able to fix it but I don't think someone wrote a clear guide for that.

zhufucdev, 2022-02-26:
The issue seem to be solved after I've done the following:
- Update to the latest BIOS
- Reset BIOS settings to default
- Disable android_init_fix
- Always connect to a secondary monitor (not likely the root cause)
There is slim chance that Android fails to initialize again, but that's not disturbing anymore