sebanc/brunch #1439: grub2win - change location of chromeos.img in windows + chromeos dualboot

dustojnikhummer, 2022-02-13:
Because I run Windows 10 and not Windows 11, I could not use wsl --mount

Which means I can not install the .img onto external storage (in my case a micro SD card)

I know the Windows drive letter has no effect on where Grub2win points, what config files do I need to change to point to a different partition/disk?

Mou786Aha, 2022-06-19:
> Because I run Windows 10 and not Windows 11, I could not use wsl --mount
>
> Which means I can not install the .img onto external storage (in my case a micro SD card)
>
> I know the Windows drive letter [...]

If you mean 'changing the location of the ChromeOS file'. Simply move it to the location you want and edit the config file in the grub2win entry. Replace the path to your new path. It's simple. 
I didn't understand one thing: If your windows 10 doesn't support wsl, how could you create an image for ChromeOS?

tomsik-radek, 2022-06-19:
> If you mean 'changing the location of the ChromeOS file'. Simply move it to the location you want and edit the config file in the grub2win entry. Replace the path to your new path. It's simple. I [...]

It supports WSL. But wsl --mount is Windows 11 exclusive. Or rather, Win10 Insider and higher.

And changing the path is not really possible either because the grub config doesn't contain the full path. It gets half of its path from a variable somewhere. Which one I don't know.