sebanc/brunch #1440: Chuwi Hi 10Go Intel n4500 no sound no screen auto-rotation....

DeadPo01, 2022-02-14:
Managed to start ChomeOS dedede 97 with brunch 97 and 5.15 kernel on Chuwi Hi 10 Go, but have issues... =(
1. No sound (alsa no sound card in verbose boot log)
2. Power button works fine on start page (login to account or choose guest user), but not working in system. ACPI_power_button enables power dialog in system on button press, but no sleep mode =(
3. No screen auto-rotation. Manual rotation in settings works fine.
4. Bluetooth no searching devices

Is it possible to fix something? Or just wait for kernel updates in brunch?

userosos, 2022-02-16:
Can you upload an image last build for use it from luve-usb on the tablet?

DeadPo01, 2022-02-16:
Didn't understand you... I've got an image from here https://cros.tech/device/dedede
And installed it on USB-flash with brunch 97.

tejasraman, 2022-02-17:
dedede? Try rammus

tejasraman, 2022-02-17:
or octopus. Try octopus. You have Geminilake, dedede is a Jasperlake image.

DeadPo01, 2022-02-17:
No. Celeron n4500 it's JasperLake, same as n5100.

tejasraman, 2022-02-17:
Oh. I was looking at an n4120 model. Try volteer.

DeadPo01, 2022-02-17:
Tried already https://github.com/sebanc/brunch/issues/1326 . No new version to try yet.

tejasraman, 2022-02-17:
Run sudo lspci int terminal

DeadPo01, 2022-02-17:
Hmm i have no terminal in OS =(

DeadPo01, 2022-02-17:
Ok, entered under account and found a lot more functions =) including terminal.

DeadPo01, 2022-02-17:
New problem... I have no enough free space to install LinuxVM to start terminal... ChromeOS insatlled on USB Flash 64Gb. How to extend user partition?

DeadPo01, 2022-02-17:
Here is the output. 
```
00:00.0 Host bridge: Intel Corporation Device 4e22
00:02.0 VGA compatible controller: Intel Corporation Device 4e55 (rev 01)
00:04.0 Signal processing controller: Intel Corporation Device 4e03
00:08.0 System peripheral: Intel Corporation Device 4e11
00:14.0 USB controller: Intel Corporation Device 4ded (rev 01)
00:14.2 RAM memory: Intel Corporation Device 4def (rev 01)
00:15.0 Serial bus controller [0c80]: Intel Corporation Device 4de8 (rev 01)
00:15.1 Serial bus controller [0c80]: Intel Corporation Device 4de9 (rev 01)
00:15.2 Serial bus controller [0c80]: Intel Corporation Device 4dea (rev 01)
00:15.3 Serial bus controller [0c80]: Intel Corporation Device 4deb (rev 01)
00:16.0 Communication controller: Intel Corporation Device 4de0 (rev 01)
00:17.0 SATA controller: Intel Corporation Device 4dd3 (rev 01)
00:19.0 Serial bus controller [0c80]: Intel Corporation Device 4dc5 (rev 01)
00:1a.0 SD Host controller: Intel Corporation Device 4dc4 (rev 01)
00:1c.0 PCI bridge: Intel Corporation Device 4dbc (rev 01)
00:1e.0 Communication controller: Intel Corporation Device 4da8 (rev 01)
00:1e.3 Serial bus controller [0c80]: Intel Corporation Device 4dab (rev 01)
00:1f.0 ISA bridge: Intel Corporation Device 4d87 (rev 01)
00:1f.3 Multimedia audio controller: Intel Corporation Device 4dc8 (rev 01)
00:1f.4 SMBus: Intel Corporation Device 4da3 (rev 01)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Device 4da4 (rev 01)
01:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device c821
```

I think there is no driver for sound... It's strange because it's integrated into intel chipset, i think

tejasraman, 2022-02-17:
I mean crosh. On desktop, with no windows open, press Ctrl+Alt+T, and at the crosh, type “shell”. Linux(Crostini) cannot look at hardware via lspci

tejasraman, 2022-02-17:
Looked online, enable the snd_hda_intel option in the settings. When you see the selection, press the down key to select ChromeOS(Settings) and press enter. Press space on snd_hda_intel and reds enter to get through the prompts(highlight kernel 5.10 and press enter) and reboot. Sound should(at least) work.

sebanc, 2022-02-17:
What is the output of "cras" command in crosh shell (without options enabled) ?

sebanc, 2022-02-19:
Those errors are actually harmless as far as I can tell, your device most likely have an issue with the SOF driver. You can try to add "snd-intel-dspcfg.dsp_driver=1" as kernel parameter.

Otherwise sound should work in kernel 5.4 I guess ?

If not, does sound work with rammus recovery image ? I did not run any test with "dedede" recovery image

userosos, 2022-02-20:
> Didn't understand you... I've got an image from here https://cros.tech/device/dedede
> And installed it on USB-flash with brunch 97.

How? I can download only recovery image. I replaced the img (used balenaEtcher) into sdcard but in bios i cant seethe sd-card like boot device.

tejasraman, 2022-02-20:
No balenaEtcher, use the ChromeOS-install script from the brunch zip file

userosos, 2022-02-20:
> No balenaEtcher, use the ChromeOS-install script from the brunch zip file

But i can see bin file. I just rename it to img

tejasraman, 2022-02-20:
But Brunch needs to do special patching to work. Download Brunch and use the script. Not Etcher.

DeadPo01, 2022-02-21:
> > No balenaEtcher, use the ChromeOS-install script from the brunch zip file
> 
> But i can see bin file in zip. I just unzip and rename bin it to img I use win 11. Will be This script works under win 11?

You should read manual about how to install brunch https://github.com/sebanc/brunch/blob/master/install-with-windows.md .
No this script will not work under Win11, it needs linux shell... So you need linux or WSL2 in windows.
This thread is not about installing brunch! There is a manual for that questions.

DeadPo01, 2022-02-21:
> 

snd-intel-dspcfg.dsp_driver=1 - added in settings before boot, nothing changed =(
Can't try kernel 5.4 - because it stucks on boot.

DeadPo01, 2022-02-21:
Tested rammus - everything the same =(( Any suggestions?

tejasraman, 2022-02-22:
Haven’t you read the manual? It says use WSL2.

userosos, 2022-02-22:
> Haven’t you read the manual? It says use WSL2.

I created an "new" issuse https://github.com/sebanc/brunch/issues/1286