sebanc/brunch #1454: Suggestion - Make brunch use of the new Chrome os flex image instead of rammus

NSbuilder, 2022-02-19:
Chrome os flex image is more suitable for most computers it seems. Brunch could make use of it by using this new image and adding features like android apps to it.

RickoDean, 2022-02-20:
I can agree with this but "WE" have to take into consideration the time it takes to compile brunch, better yet: brunch intergrated and running on google flex image. My brunch broke on me at update r98 so I'm using FLex. I hate that it doesn't have android nor linux support bur I'm getting by running things in the browser. I miss brunch but it's no longer running correctly for me, using the appropriate rammus/chrome/brunch version. it's a good idea but I'm sure that it will take some time for something useful is ready on this magnitude.

Joshagos, 2022-02-21:
Great idea! I completely agree with you.

WesBosch, 2022-03-01:
Brunch gets android support from the recovery, it doesn't add android. Using flex in its current state (The Reven recovery) would mean having less hardware compatibility and no android.

Joshagos, 2022-03-01:
> Brunch gets android support from the recovery, it doesn't add android. Using flex in its current state (The Reven recovery) would mean having less hardware compatibility and no android.
Maybe we should wait until it gains Android support then? It isn't even in the Stable channel yet...

WesBosch, 2022-03-01:
Even then, it'll be treated like any other recovery. Right now we are not suggesting Reven over any of the other recoveries as it's only a net loss in features.

Joshagos, 2022-03-02:
> Even then, it'll be treated like any other recovery. Right now we are not suggesting Reven over any of the other recoveries as it's only a net loss in features.

For now, yes. I hope that'll change soon though!

sortedcord, 2022-04-10:
> Even then, it'll be treated like any other recovery. Right now we are not suggesting Reven over any of the other recoveries as it's only a net loss in features.

Is it possible to use a flex recovery image along with brunch? My use case doesn't really require android apps so I'm fine with its exclusion in the os. Flex doesn't support my Wlan card (rtl8723de) along with a few other things but brunch does...

WesBosch, 2022-04-10:
It is possible to use the Reven image, but there is no advantage to doing
so and support is generally worse than just using a standard recovery.

I would strongly suggest using a normal recovery, not flex. If you don't
want to use android, it can be disabled from the settings page.

dragon788, 2022-06-11:
I think a big issue with ChromeOS Flex aka CloudReady is that on older hardware you NEED microcode updates to use Linux apps, and Google isn't prepared to add those updates for ALL the possible models people will install CrOSFlex on, while Brunch DOES inject those updates, allowing you to use Linux apps on new or older systems.

So a BrunchFlex combo would be amazing for people who don't want/need Android apps (they tend to be battery hogs whether using the older ArcVM or the newer container method).

sortedcord, 2022-06-27:
Indeed, I am using it on a daily basis without much of an issue. Although my laptop isn't that old, the main reason why I am using it because brunch supports my laptop's wlan card but CrOSFlex didn't until recently as well as other crucial fixes like internal mic and 3+ finger gestures

lavilao, 2022-07-19:
Hi I have 2 questions, I currently have this image ```chromeos_14794.0.0_reven_recovery_dev-channel_mp-v2.bin.zip``` wich is based on chromeos 103 my question is when brunch 103 hits stable (currently on unstable) can I use it to dual boot linux and chrome os flex? and my second one is given that chrome os flex already has broad hardware support baked in (thus not needing brunch custom kernel or firmware) can I enable automatic updates in it (Chromeos flex)? Thanks in advice

sebanc, 2022-07-22:
I never really looked at this thread until now, to clarify a few things:
- brunch apparently works with ChromeOS flex "reven" recovery image (according to reports, I never tried it).
- it might potentially fix some issues with crostini not installing in ChromeOS flex (due to a kernel patch).
- However, it will be less stable than pure ChromeOS flex and you wil need to keep updating brunch regularly.

lavilao, 2022-07-23:
Ok, thanks for answer so un terms of stability chromeos is better. Also in My case Flex gestures did not worked yet on chromeos they did.

b4iterdev, 2023-02-24:
I think the only reason for running chromeos flex on brunch is for having dualboot capability, but it seems that chromeOS Flex have less bug with my device as the current issue that I found is that crOS on brunch always report discharging while crOS Flex does not have this issue

sebanc, 2023-02-24:
If ChromeOS Flex supports your hardwar and you just want to dualboot it in an image (using the same principle as brunch) you should use this:
https://github.com/sebanc/linuxloops

The only reason to use brunch with flex would be if you have drivers that are supported by brunch and not by flex.

madhits, 2024-04-15:
I have a question related to this, hopefully @sebanc can answer. Why is it that cloudready/Flex can support devices with only SSE3/ssse3 and brunch requires SSE4? Is this to do with your rootc.img customization? What in brunch requires sse4? Is it android support? I understand if you just had to make this easier on your build process. Cloudready/flex does not have both virtualization environments needed for linux and android and my guess is those have something to do with sse4. Could brunch just allow the install on SSE3 systems and have android/linux not work?

lastly in there are chromebooks with only sse3, such as edgar, coral, and a few others, mostly AUE models now. So if those models exsist and can be used with brunch then essentially is brunch holding people back from using a recovery with everything they need as in CPU/instructions support but brunch installers rootc.img puts in place a restriction that otherwise would not exist for them using that recovery. I know this would basically turn brunch into a direct competitor to flex and maybe you dont want that or google told you you cant do that.

sebanc, 2024-04-16:
ChromeOS devices images are built using specific board overlays:
https://chromium.googlesource.com/chromiumos/overlays/board-overlays/+/refs/heads/main

Depending on the cpu Gen, ChromeOS devices images will be generated using a specific config for example for a Skylake gen cpu:
https://chromium.googlesource.com/chromiumos/overlays/board-overlays/+/refs/heads/main/chipset-skl/profiles/base/make.defaults
You will find:
```
BOARD_COMPILER_FLAGS="-march=silvermont -maes"
BOARD_RUSTFLAGS="-Ctarget-cpu=silvermont -Ctarget-feature=+aes"
USE="${USE} march_silvermont"
```

ChromeOS-Flex images are just built using a generic overlay that has low requirements in terms of cpu instructions.

This has nothing to do with brunch.

closed 2024-04-16 by sebanc