sebanc/brunch #1455: Speaker didn't work on my AMD Ryzen Device

H2Aoi, 2022-02-20:
the speaker didn't work.When I connect my PC to another display using HDMI,the display's speaker works well.i guess it may be some driver problems.
i'm using AMD Ryzen 5 3500U,and my speaker is realtek

xtibrox, 2022-03-01:
> the speaker didn't work.When I connect my PC to another display using HDMI,the display's speaker works well.i guess it may be some driver problems. i'm using AMD Ryzen 5 3500U,and my speaker is realtek

hi what chrome image and Brunch release you are using?i have amd ryzen 5 to but i cant install chrome os

H2Aoi, 2022-03-02:
> hi what chrome image and Brunch release you are using?i have amd ryzen 5 to but i cant install chrome os

i use the latest brunch release and zork,it works well

xtibrox, 2022-03-02:
> i use the latest brunch release and zork,it works well

I have AMD Ryzen™ 5 4600G Processor ai try with brunch 97 and zork image and i Stuck at "Patches are being applied"i wait 1hour and nothing

xtibrox, 2022-03-02:
> > > > > the speaker didn't work.When I connect my PC to another display using HDMI,the display's speaker works well.i guess it may be some driver problems. i'm using AMD Ryzen 5 3500U,and my speaker is realtek
> > > > 
> > > > 
> > > > hi what chrome image and Brunch release you are using?i have amd ryzen 5 to but i cant install chrome os
> > > 
> > > 
> > > i use the latest brunch release and zork,it works well
> > 
> > 
> > I have AMD Ryzen™ 5 4600G Processor ai try with brunch 97 and zork image and i Stuck at "Patches are being applied"i wait 1hour and nothing
> 
> do you notice that [Ryzen 4xxx devices need kernel 5.10](https://github.com/sebanc/brunch/blob/master/troubleshooting-and-faqs.md#kernels)?

how to do this please help me