sebanc/brunch #1456: After a reboot, all my Android apps are just spinning.

dexteruseslinux, 2022-02-20:
Every one of them. Before the reboot, every one I had installed (Twitch, Opera GX, Parsec, Sketchbook) had worked however Roblox hadn’t.

Keep in mind I’m using a ASUS Chromebox CN60 modified with MrChromeBox’s firmware to run Brunch. Mine has a Intel Celeron 2955U, 4GB RAM (I tried 8GB but it won’t recognize past 4 on mine for some reason), and a 16GB mSATA SSD.

How could I fix the issue with Roblox and other Android apps? I have plenty of other devices I can play that on and get no lag on, but I wanted to try it out on this Chromebox too.

dreimer1986, 2022-02-23:
Confirmed the same behavior on my Acer Aspire 5755G with r98 rammus and r97 brunch. Before reboot things were fine, after things were spinning and cannot open.

sandyponytail, 2022-02-25:
I have the same issue.  I have two brunch computers.  The first one is J1800 cpu, r86 for both rammus and brunch, and it worked perfectly for one year including Google Play.  But yesterday, all of a sudden, all of the Android apps installed were unable to load but keep spinning.  I originally thought that the OS was outdated or something error after one year of running.

However, I just install r97 (both rammus and brunch) to a new computer (N5095 cpu) one hour ago and I found the same issue occurred after the new installation.  Actually I used the same image, same USB to install the r97 to another computer (N3700 cpu) one month ago and the Google play works without problem at that time so I don't think the problem was caused by the OS image.  Unfortunately the N3700 computer was dead so I cannot try if the Google play still work.

I just tried "android_init_fix" on the N5095 computer but the issue has not solved.

Due to the problem occurred in different version of Chrome OS (r86 and r97), I suspect there is some update for Google play caused the issue.

Please advise how to fix the issue.  Thank you.

RickoDean, 2022-02-26:
Same issue here Lenovo Thinkpad T430 128GB SanDisk mSata SSD Card, 8GB Ram. Rammus updated to 98 from 94 & I upgraded my brunch from r94 to r97 before I updated to 98. If I'm not mistaken, do brunch and rammus have to be the same version? Granted I did encounter android app issues before i ran updates

username0occupied, 2022-03-30:
The same issue on my laptop,hp elite x2 1012 g2 r96 rammus and r97 brunch.The apps only run in the first launch after installing the system,they all spin after the system restarts.

username0occupied, 2022-04-11:
I have solved it.You can get help from https://github.com/sebanc/brunch/issues/74