sebanc/brunch #1462: rammus ver. 98 - CTRL + ALT + T not working

artemnext, 2022-02-22:
I'm install rammus 96 image. All work perfectly.
I trying to install 98 image. Making USB, boot from it, and when i try start terminal to install chrome os on hdd. Terminal blink and close window.
I try to install 96 and update brunch and chrome os to 98 and everything happened again.

artemnext, 2022-02-22:
ASUS H61M-K MB, i3-2100, RAM 4GB

RickoDean, 2022-02-26:
I've encountered the same problem. I installed r94 (brunch+rammus) it worked fine for a couple days then my android apps begain bugging out and wouldn't open. I upgraded to 98 (from brunch 97) and my crosh won't work and my android apps won't open

Lenovo Thinkpad T430
Intel Core i5 vPro
8 GB Ram
128 GB mSata SanDisk SSD card

rlaunemacedo, 2022-03-22:
Instalei em meu notebook HP Pavilion G4 a versão 97
https://rlaunemacedo.github.io/

Quando fiz a atualização para o 98, o crosh deixou de funcionar... tive que re-instalar o 97 de volta!

PicPew00, 2022-03-29:
i have the same problem also i have legacy and used mbr method.i made some experiments and from version 93-97 alt ctrl t is working but you cant install linux. at version 98 ctrl alt t is crashing and also you cant install linux. version 90 and 92 both works.