sebanc/brunch #1469: Float Question: How Likely Playstore will work to any device?

revdeluxe, 2022-02-26:
This isn't really a issue but for those who stumble upon my crappy post.

out of 10 brands of budget computers/laptop.

How likely is it going to run. either will it run, little hiccups, or impossible.

I'm currently(this summer 2022) going to try brunch(should i say chrome os.). to my HP Intel Celeron n3060.

elder5, 2022-04-25:
How did it go?
I ordered a chromebook for someone else. same one with same processor. But it's support ends on June 2022...so my fix would be to go with brunch, no expiration date on the OS.
My main device is a dell venue 11 with brunch OS, I love it!

revdeluxe, 2022-04-26:
> How did it go? I ordered a chromebook for someone else. same one with same processor. But it's support ends on June 2022...so my fix would be to go with brunch, no expiration date on the OS. My main device is a dell venue 11 with brunch OS, I love it!

I'd said that i was not successful. but i was able to boot into the cloudready installer. but it keeps crashing so i moved to Archcraft instead.

closed 2022-04-26 by revdeluxe

elder5, 2022-04-28:
Update:
I have inform that I successfully bypass Google "AUE"  (auto update expiration) using brunch on the 3189!!
I used "windows install helper - coolstar. Org" tutorials. 
Then Ubuntu live usb to install brunch. 

[image]
[image]