sebanc/brunch #1473: version `GLIBC_2.33' not found (required by /usr/bin/coreutils)

weberk, 2022-02-28:
Hi, just updated to ChromeOS 98.0.4758.107 and everything looked fine until I started a shell via crosh.

Every command claimed:
chronos@localhost /bin $ ./ls  
/usr/bin/coreutils: /usr/local/lib64/libc.so.6: version `GLIBC_2.33' not found (required by /usr/bin/coreutils)

Any chance to recover from this state?

ZZZXXX0110, 2022-03-04:
Having the same problem also, with r98 rammus. Could not even start the shell with the "shell" command after opening the crosh window with Ctrl+Alt+T.

ZZZXXX0110, 2022-03-04:
But it seems like a lot of people running the official Chrome OS on official Chromebook devices are getting the same problem on r98, so this might not be a problem with Brunch but rather with Chrome OS itself. No idea what Google did to F up this on their end lol