sebanc/brunch #1479: RTL8821CE, PWA, or Any Options Won't Load in Zork's Chrome OS

novalkomik, 2022-03-04:
I've been using Chrome OS with Brunch on my Zork (HP-CM14 series with AMD Ryzen 5 2500U processor) for almost a week and I clearly happy with the performance.

Unfortunately, I've realized that my `rtl8821ce` Realtek WLAN chip won't load properly. I'm relying on my Ethernet connection to access the Internet. It would be okay for now, unlesss I have to go outside and carry my device.

Before made the decision to push this Issue, I've been searching on the open and also closed issues such as **[this](https://github.com/sebanc/brunch/issues/584)**, **[this](https://github.com/sebanc/brunch/issues/444)**, **[this](https://github.com/sebanc/brunch/issues/1402)**, and **[this](https://github.com/sebanc/brunch/issues/1024)**.

All of those issues have some critical yet informative tips, such as:

> 1. You've might be using dualboot between CrOS and Windows/Linux (Therefore you need to manually edit the grub-config)
> 2. The Bluetooth is working, but not the WLAN
> 3. Someone are able to fix it, but only in terms of activating the WLAN and unable to connect (After re-read the whole discussion, the person who's thought he's using RTL8821CE is actually using Broadcom)

So what I've been doing to atleast fix the issue is by editing my GRUB configuration thru Windows 11 by using Grub2Win. Adding the `options=opt1,opt2,...` after the `cros-debug` and then applying the config. I've also tried to put PWA and ALS activated thru `options=`, but unfortunately these options were somehow ignored.

 I've printed my dmesg, and It actually execute the configuration:
`[    0.000000] Command line: BOOT_IMAGE=(loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug option=pwa,rtl8821ce loop.max_part=16 img_part=/dev/sdb6 img_path=/chromeos.img`

But the question is, the lspci said the other thing:
`	Kernel driver in use: r8169
	Kernel modules: r8169`

So, by all the things that I've tried (and also the txt that I put below), what step did I missed here? After looking the similiar RTL8821CE issues, what is actually wrong inside the configuration and what should we do? Did brunch didn't save the config? Did the options were actually not compatible at all with the RTL8821CE issue?

Attachment: **[Dmesg.txt](https://raw.githubusercontent.com/novalkomik/txt-in-need/main/dmesg.txt)** | **[Lspci.txt](https://raw.githubusercontent.com/novalkomik/txt-in-need/main/lspci.txt)**

sebanc, 2022-03-04:
I don't support anymore custom grub configs as there is a built in configuration menu which does all of this properly but since you took the time to post a lot of information and the logs, I will point out that you used "option=" in the grub config instead of "options="

novalkomik, 2022-03-04:
> I don't support anymore custom grub configs as there is a built in configuration menu which does all of this properly but since you took the time to post a lot of information and the logs, I will point out that you used "option=" in the grub config instead of "options="

I'm deeply embarrassed right now due to the mistype. As I'm replying, I'm already fix the typo I made. Is it fine for the OS to took a long `booting in blind mode` after changing the `options=`?

sebanc, 2022-03-04:
No, you probably broke the grub config somehow. Make a clean install of brunch using the github guide and use the config menu this time

novalkomik, 2022-03-04:
> No, you probably broke the grub config somehow. Make a clean install of brunch using the github guide and use the config menu this time

Will do. And yes, apparently It's broken. It goes whitescreen for half an hour after I quote-reply you.

novalkomik, 2022-03-05:
I tried the clean install and apparently both of the options (the Options and the normal one on grub) won't even load the OS.
It stuck for nearly 5+ hours (while I'm sleeping). Should I need to clean install it again and give you more details about that?

novalkomik, 2022-03-05:
Also, I don't know precisely but, is that okay for Grub2Win to send this warning after I put the config?

[image]

rlaunemacedo, 2022-03-22:
Instalei em meu notebook HP Pavilion G4 com placa Ralink RT3290 Wireless e para fazer funcionar o bluetooth eu adicionei a rtbth a options, conforme listagem abaixo.

rmmod tpm
img_part=/dev/sda7
img_path=/chromeos.img
search --no-floppy --set=root --file $img_path
loopback loop $img_path
linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off  \
cros_secure cros_debug options=rtbth,enable_updates loop.max_part=16 img_part=$img_part img_path=$img_path \
console= vt.global_cursor_default=0 brunch_bootsplash=default
initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img

Talvez isso possa ajudar em algo.