sebanc/brunch #1480: Dual Boot on Surface Go 2, Boot partition not found

Dragonsky8, 2022-03-05:
Hello,

When booting into ChromeOS via grub, my surface remains stuck on loading brunch framework, after which it proceeds to restart. It remains stuck there for around 10 seconds, before restarting.

When I tried to see into the ChomeOS(settings), the final automated response is "the boot partition was not found, falling back to shell..."

I'm not sure what's causing this, as I presume the Grub did find the Chromeos.img file on my drive.

Kadoyu, 2022-03-30:
You may have encountered the same problem🤔

makingitup, 2022-04-20:
I'm also having this problem.
2021 Hp spectre x360

codelif, 2023-02-02:
@makingitup 
Check if you have RAID mode enabled, if so set it to AHCI mode. 
Doing so will fix your Chrome OS but Windows will not boot.

To fix windows, check this superuser thread:
https://superuser.com/a/1359471