sebanc/brunch #1481: Black screen after chrome logo

rohitojha484, 2022-03-06:
I have asus laptop
Cpu- ryzen 5 3550h
Gpu - radeon rx560x
I tried installing chrome os via bootable usb and grub2win , but in both the case my screen becomes black after the chrome logo, i have tried different versions of brunch and zork recovery

ZaK-bg, 2022-03-08:
rx 580 8gb black screen

neptalactone, 2022-03-20:
You have the same hardware as my laptop. and both FydeOS and ChromeOS with brunch have the same result. I think it's due to our laptop have 2 AMD GPUs. the iGPU one (usually vega) and the dGPU one. I have a feeling that this has something to do with dgpu.sh patch that gets executed by brunch. 

I am not a programmer so i can't look further. it's just my assumption. If you try ChromeOS Flex, you'll see that you booted into it fine. And it will use the right GPU (the integrated graphics one). If your laptop has an option in BIOS that lets you disable your dGPU (dedicated GPU) which in this case RX560X and RX580, i suggest you try to disable it first.

If you don't have that option, try to insert the "fbcon=map:1" custom kernel parameters through ChromeOS settings menu. If you can understand bash scripting and basic programming, try to look into the dgpu.sh patch in this repository. 

Our only option is if @sebanc replies to us and maybe makes a special patch for devices that have 2 AMD GPUs so that it will only use the integrated one like ChromeOS Flex. Or you can just use ChromeOS Flex, which features only basic ChromeOS. no android, no linux.

lentas1, 2022-06-14:
chrome os flex haves linux

neptalactone, 2022-06-26:
Update: It seems ChromeOS and brunch version r101 can now boot my specific hardware like ChromeOS Flex.

So if you have hardware that like this:
CPU: AMD Ryzen 5 3550H
iGPU: Vega 8
dGPU: RX 550X

Specifically this device: Acer Nitro 5 AN515-43.

It will now can boot to ChromeOS r101 with both brunch version r101 and r102 just fine. With Acer Nitro 5 all working great. With a little caveat that if you do something to the android system that make it won't open, you have to powerwash it. (like update from r101 to r102 in ChromeOS settings, will make android app won't open unless you do powerwash)