sebanc/brunch #1483: Black screen a few seconds after first boot?

v-mod, 2022-03-09:
When I boot it goes on a menu with the options **ChromeOS** and **ChromeOS settings** but it then disappears and I get a black screen after a few seconds. Does anyone what the issue might be?

v-mod, 2022-03-09:
I installed it on a USB drive using wsl2 with the instruction provided in install_with_windows.md

ouija, 2022-04-03:
After a few seconds, grub will automatically boot one of the selections if none were picked.   Did you update the options and set the bootsplash screen to be "blank" and aren't seeing the brunch loading screen because of this?

v-mod, 2022-04-04:
I think i may have found the issue. The PC I am using has a x64 processor but only 32-bit os will boot on it. After a bit of reading I think it has a 32-bit uefi and to my knowledge bruch isn't compatible with 32-bit.