sebanc/brunch #1484: Can't open playstore if add anything option in R97 Chell ver. 97 stable, 98 stable & 101 dev

panduwinata1996, 2022-03-10:
**Playstore** can't opening if brunch R97 setting _update/change_.
i try to add/select this option **"android_init_fix"** on **sudo edit-brunch-config** but **not fixing** this issue.

jhuny-arj, 2022-03-12:
@panduwinata1996
Hi, Do you use chromeBrew on your system?
  Maybe it's a coincidence, but when I installed chromeBrew the android apps started to crash. It wasn't immediately, it was gradually, and after I restarted the PC, I couldn't use android apps anymore. Only when I reinstalled chromeOS/Brunch, the apps started working again.
  I'm also trying to identify what causes these android apps crashes.

panduwinata1996, 2022-03-13:
no, im in fresh installed position and i try find this problem but nothing solved about this issue.
i think, this issue about brunch r97 not full inject kernel support to chell recovery.

This problem if we running sudo edit-brunch-config.
But if we not running _sudo edit-brunch-config_, overoll no problem/normaly, but low performance for my laptop.

this stuck problem not only android vm but linux beta too

helpme9, 2022-04-03:
hi,i am using r98 zork, i am getting problem, my playstore is not work/opening and show error-7 and i can't install any apk.........

z2168235, 2022-04-15:
I'm having same problem playstore not working. I'm using r100 cyan.
Older version r91 worked fine.