sebanc/brunch #1485: Chrome OS crashes randomly in triple-boot on all channels Rammus

crimsonvanitas, 2022-03-12:
So, I recently installed Brunch to my Lenovo Thinkpad Yoga 11e in a triple boot. I recently upgraded my SSD and have been having some issues with it crashing unexpectedly when downloading any sort of file, whether it is from Drive or through the Chrome Web Store. I tried to download an adblocker today and it crashed. I've heard of other people having this issue, but they said it was solved back in v81 or 83. I'm a bit tech-savvy myself and this has thoroughly stumped me. I'm on Rammus, version 101 on dev channel.

jhuny-arj, 2022-03-12:
Look, even though it's not a requirement...
The "recommended" is to leave it in the same version as brunch
https://github.com/sebanc/brunch/blob/master/troubleshooting-and-faqs.md
I did the same before and was forced to downgrade to r97+r97 ("stable").
It is very common in beta/dev/unstable/nightly projects to have stability issues. I think that's the reason developers make a separation between stable, beta, alpha or DEV... Try a Downgrade! in my situation it solved the problem.

crimsonvanitas, 2022-03-12:
Ok, thank you for the help!

rlaunemacedo, 2022-03-25:
Instalei em meu notebook HP Pavilion G4 com placa Ralink RT3290 Wireless e para fazer funcionar o bluetooth eu adicionei a rtbth a options, conforme listagem abaixo.
Tenho Linux Mint, Windows e ChromeOS em Tri-boot funcionando. Fiz a instalação pelo Linux.
https://rlaunemacedo.github.io/

rmmod tpm
img_part=/dev/sda7
img_path=/chromeos.img
search --no-floppy --set=root --file $img_path
loopback loop $img_path
linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off
cros_secure cros_debug options=rtbth,enable_updates loop.max_part=16 img_part=$img_part img_path=$img_path
console= vt.global_cursor_default=0 brunch_bootsplash=default
initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img

Talvez isso possa ajudar em algo.

closed 2022-04-02 by crimsonvanitas