sebanc/brunch #1486: Dual Boot Issue

wadkarkundan, 2022-03-12:
I want to dual boot ChromeOS with Win11 in 3gen 
Now the main issue I have a 1TB hardisk where I have  created 2 partitions, 1 on which win11 is installed and the other 256GB I want for ChromeOs. 
So I'm using this commands on Windows WSL Linux(Ubuntu) 
Version- Samus 91 and brunch91
★ Commands ★ 
sudo apt update && sudo apt install pv tar cgpt 
cd /mnt/c/users/*YOUR USERNAME*/*FILE LOCATION* 
ls 
sudo tar zxvf *REPLACE WITH YOUR BRUNCH VERSION* 
sudo bash chromeos-install.sh -src *REPLACE WITH  
YOUR CHROME OS VERSION* -dst /mnt/ 
*PARTITION Letter*/*YOUR FOLDER NAME* -s *image size*

But when I execute the command sudo bash chromeos-install........ I get an error saying dd failed: is a directory? and no I'm writing everything correctly the filename isn't the issue. I'm attaching the Screenshot Shot if the error can u throw some light on this. I wasted my whole entire day figuring this out.
[image]

jhuny-arj, 2022-03-12:
To generate the Chrome OS image, it is necessary to put the destination directory and "chrome.img", if you put only "chrome": it will be identified as a folder... Then the error appears.

Wrong Way: "/mnt/f/chrome"
Correct Way: "/mnt/f/chrome/chrome.img"

wadkarkundan, 2022-03-13:
> To generate the Chrome OS image, it is necessary to put the destination directory and "chrome.img", if you put only "chrome": it will be identified as a folder... Then the error appears.
> 
> Wrong Way: "/mnt/f/chrome" Correct Way: "/mnt/f/chrome/chrome.img"
> 
> #### sudo bash chromeos-install.sh -src chromeos_filename.bin -dst ~/tmpmount/chromeos.img -s size

This is where I followed it from and the same command is working there which I'm trying to execute
https://youtu.be/IUuJIkPssEs

jhuny-arj, 2022-03-13:
I saw the video, it's like you said.  maybe the script version affected, it seems that in the video he used it to generate r94 image.  did you test it the other way, the way it is in the official guide available right here on git, with chrome.img instead of just the path to the folder?

wadkarkundan, 2022-03-14:
Well too late now to check unfortunately after trying a lot and struggling installed Linux as dual boot. But thanks for your help anyways if someone comes across this issue please check the solution provided and confirm its working

closed 2022-03-14 by wadkarkundan