sebanc/brunch #1490: RTL8723BS SDIO Wifi not working on iWork10 Tablet

clee604, 2022-03-22:
First of all, I'd like to thank all the contributors for this, I've installed Brunch on several machines and it is great to reduce ewaste in keeping some old devices usable.

I have an issue with a tablet (iWork10) that has a Cherry Trail processor. Everything works fine running the latest Brunch and Rammus recovery on the default kernel except for 2 things: 
1. the touchscreen is inverted on the x-axis (I fixed this using of_touchscreen option found in a previous issue on git)
2. the wifi will not connect (it does work in windows, so it's not a hardware issue, however I did boot into FydeOS using a live usb and it also has same issue; I think remember the wifi working in Linux Mint though).

This issue is regarding the wifi which is provided by a RTL8723BS SDIO adapter which is a wifi/BT combo. Bluetooth works fine, and wifi detects networks fine but when trying to connect, it gives the following error:
"Network connection error, Failed to connect to network: Unrecognized error: unknown-failure."

Things I've tried: 
- changing to the 2 newer kernels
- changing to the unstable brunch
- different wifi router, using my phone as a hotspot, with and without authentication
All exhibit same connection error.

In my googling, I saw that there is a reference to the RTL8723BS driver in R94 release notes (https://github.com/sebanc/brunch/releases/tag/r94-stable-20211121) so I assume this device *should* be supported.

sudo lspci doesn't list the device, as I assume SDIO adapters use a different interface.
00:00.0 Host bridge: Intel Corporation Device 2280 (rev 22)
00:02.0 VGA compatible controller: Intel Corporation Device 22b0 (rev 22)
00:03.0 Multimedia controller: Intel Corporation Device 22b8 (rev 22)
00:0b.0 Signal processing controller: Intel Corporation Device 22dc (rev 22)
00:14.0 USB controller: Intel Corporation Device 22b5 (rev 22)
00:1a.0 Encryption controller: Intel Corporation Device 2298 (rev 22)
00:1f.0 ISA bridge: Intel Corporation Device 229c (rev 22)

Attached here is the dmesg.txt, [dmesg.txt](https://github.com/sebanc/brunch/files/8328249/dmesg.txt)
I tried to connect to the wifi right before creating the dmesg, and the last few lines at the bottom seem to point some kind of error that is beyond my knowledge. Any help or suggestions on how to resolve this would be greatly appreciated!

[  866.021604] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
[  872.937861] RTL8723BS: rtw_set_802_11_connect(wlan0)  fw_state = 0x00000008
[  872.950541] RTL8723BS: start auth
[  872.954803] RTL8723BS: auth success, start assoc
[  872.965289] RTL8723BS: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[  872.965305] RTL8723BS: assoc success
[  872.978818] RTL8723BS: send eapol packet
[  872.996731] RTL8723BS: send eapol packet
[  872.997276] RTL8723BS: set pairwise key camid:4, addr:44:d9:e7:f3:94:89, kid:0, type:AES
[  872.999229] RTL8723BS: set group key camid:5, addr:44:d9:e7:f3:94:89, kid:1, type:AES

ouija, 2022-04-02:
I am experiencing the same issues using a Toshiba Encore 2 WT8-B _(which uses a Baytrail-T dual-core SoC)_ and created a [duplicate] issue about this [here](https://github.com/sebanc/brunch/issues/1497) before seeing this one.

Previously I used an [alternative driver](https://github.com/rockchip-linux/kernel/tree/release-4.4/drivers/net/wireless/rockchip_wlan/rtl8723bs) in a build of Android-x86 on this device and found it to be much more stable and fixed issue with consistent network connection issues and drops.

clee604, 2022-04-27:
I've solved this, while I'm sure it is related to ChromeOS or the driver somehow (since wifi works in Windows and Linux Mint), I have managed to get wifi working.

For those who encounter similar issues, the problem is the PMF setting in my Unifi AP and controller for my wireless network. I had it set to "optional." Setting this this to "disabled" allowed me to connect.

closed 2022-04-27 by clee604