sebanc/brunch #1491: How to force landscape on tablet mode for surface pro 4

Xeroka88, 2022-03-24:
Hi!
First of all.. Thank you for this amazing project.!!🙌
I have few issues that i will like to ask about.

1. On tablet mode went i use an App like youtube music vanced, it auto rotatate the screen to potrait. but in desktop mode, it forced to be on landscape, it put a lot of black on the side... but well.. soomin in the app will fix that but still is better that way.
i have try:
- lock the screen from notification bar, you can lock the position, but is not working at least for this app, everytime it turns arround.
- download apps for rotation orientation, but didn work.
i thing there may be a way on the setting of brunch?? or in tsomeway by linux terminal.?

2. Camera not working. i think this is a general issue in everywhere, just wanted to know if someone had make this to work. 😊

Thanks you for your time.

leo0725, 2022-04-04:
Hi, Xeroka 
how is touch screen working on surface pro 4 now? 
i try brunch few month before, but is will sometimes no response on my pro 4, 😭

Xeroka88, 2022-04-04:
Hi. Is working  fine.. i use the kernel 5.x.x i dont remember  exactly  the number.. but is not the lastest.. i did change to another OS anyway.. because  i chrome os was to simple.. so im using now android x86 on my surface pro 4

Lucaacer, 2022-04-08:
> 
Have a look here https://github.com/sebanc/brunch/issues/550

 "options=sysfs_tablet_mode" will disable your keyboard (if attached) and

echo 0 | sudo tee /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode

or

echo 1 | sudo tee /sys/bus/platform/devices/tablet_mode_switch.0/tablet_mode

will change screen orientation