sebanc/brunch #1493: Crostini and arc++ container resetted after everytime changed brunch settings

ISNing, 2022-03-28:
I changed form native chromeos on chromebook(Lenovo Yoga Chromebook C630,pantheon,nami) to brunch by flashing UEFI firmware form mrchromebox.

I used recovery image R98 of pantheon, with flag enable_updates, pwa, native_chromebook_image, and hyper_threading force enabled.
Brunch ver.: R97
ChromeOS: R99 updated by chromeos internal updater.
Intalled in internal emmc device.

After everytime I changed brunch settings, I found that crostini and arc++ container has resetted, and also, I'v been asked to re-login my google account, and I think this shouldn't be happen.
It seemed that it's related to rebuilding rootfs?

I'm not familiar with brunch, and don't know how it work.
Is this expected to happen? And how can I prevent it before updating?

ISNing, 2022-03-28:
It's so wired but after (I didn't did anything, may be because a hibernation:?) it happened again in this morning, it did happened again at least till now...
And it finnally turned out that it's not related to rubuilding rootfs, it's just a wired bug?

ISNing, 2022-03-29:
Yes, it happened again, just after a reboot.(because after a hibernation, I found my headphone jack dectection broken, though i have pulled out my headphone, the headphone input/output can still be seen, may be this is also a signal).

After this, everything are lost, just like the first time i boot the system, but account name or something else were kept...

help needed...
I don't wan't to restore crostini and install apps by consuming many hours anymore......

ISNing, 2022-03-29:
What's more, cookies(uncertain) has gone, too.
At least I have to login in every site, which is promised that I have already logged in before the reboot.

ISNing, 2022-04-01:
I'd found the reason from the telegram group, here:
"⚠️ Please Read ⚠️ 

ChromeOS 97 has started to roll out. Due to major changes to the system, it is strongly suggested to avoid updating past 94 for stability. Expect bugs after that point. 

The recent stable update has introduced a major bug for Brunch users. There have been confirmed reports of ChromeOS 96 and higher forcing powerwashes after updating. A powerwash is a type of reset in ChromeOS that erases userdata, but leaves the rest of the system mostly intact similar to a factory reset. 

This means that your userdata may be deleted when you update! Please make sure you have backed up everything important before updating to ChromeOS 96 or higher. It is also suggested that you have the latest Brunch release installed. This issue will not be resolved so please make preparations.

Brunch settings, framework options and alternate kernels will not be reset, but certain Brunch features including the Brunch Toolkit, The Brunch PWA and Brioche containers may need to be set up again. If you intend to update, select the enable_updates framework option from the Brunch Configuration Menu using the ChromeOS (Settings) boot entry. 

Additionally, ChromeOS 96 and higher are known to have major issues with apps not loading, there is no known solution. Beta and dev channels have similar issues. For stability, it is suggested to stay on ChromeOS 94."

Maybe it would be a good idea to notice this on github...@sebanc

ISNing, 2022-04-01:
Rolled to R94, still powerwashing, and sound do not work too.

tejasraman, 2022-04-19:
Rammus image and chromebook kernel might work :)

ISNing, 2022-04-20:
> Rammus image and chromebook kernel might work :)

But it's obvious that the rammus image will be in lack of sound driver(or device specified configuration) for pantheon, which will made sound card not work.

tejasraman, 2022-04-20:
UGH you didnt read the troubleshooting page!
The chromebook_audio brunch option loads the correct sound driver from an actually native image while keeping rammus

Tejas Raman (he/him/his)

Contact me at ***@***.*** ***@***.***/0?redirect=mailto%3Atejasraman%40icloud.com&recipient=cmVwbHkrQVhEWUtEQ0FOT0dOSkFES0tTQTRZRVdBTlNFRVRFVkJOSEhFTkFFTEpVQHJlcGx5LmdpdGh1Yi5jb20%3D)
My website: tejasraman.36bit.com ***@***.***/1?redirect=https%3A%2F%2Ftejasraman.36bit.com&recipient=cmVwbHkrQVhEWUtEQ0FOT0dOSkFES0tTQTRZRVdBTlNFRVRFVkJOSEhFTkFFTEpVQHJlcGx5LmdpdGh1Yi5jb20%3D)

ISNing, 2022-04-20:
> UGH you didnt read the troubleshooting page!
> The chromebook_audio brunch option loads the correct sound driver from an actually native image while keeping rammus
> 
> Tejas Raman (he/him/his)
> 
> Contact me at ***@***.*** ***@***.***/0?redirect=mailto%3Atejasraman%40icloud.com&recipient=cmVwbHkrQVhEWUtEQ0FOT0dOSkFES0tTQTRZRVdBTlNFRVRFVkJOSEhFTkFFTEpVQHJlcGx5LmdpdGh1Yi5jb20%3D)
> My website: tejasraman.36bit.com ***@***.***/1?redirect=https%3A%2F%2Ftejasraman.36bit.com&recipient=cmVwbHkrQVhEWUtEQ0FOT0dOSkFES0tTQTRZRVdBTlNFRVRFVkJOSEhFTkFFTEpVQHJlcGx5LmdpdGh1Yi5jb20%3D)
> 
> >
> >
> > > Rammus image and chromebook kernel might work :)
> > But it's obvious that the rammus image will be in lack of sound driver(or device specified configuration) for pantheon, which will made sound card not work.

Soory, I have already tried it in image for pantheon, if I turn that flag on, then the sound no longer working, but it's working before...

tejasraman, 2022-04-20:
That was a poorly rendered email signature. You can reply by email you know :) but I didn't this time.

Maybe using the generalized nami image and then the flag will pull from the pantheon audio codec/driver

closed 2024-02-19 by ISNing