sebanc/brunch #1497: Alternate (more stable) driver for the RTL8723BS

ouija, 2022-04-02:
Hi there,

I've tested brunch on this Toshiba Encore 2 WT8-B device of mine, and found that the RTL8723BS driver is suffering from random disconnects and issues caused by the driver, and wanted to mention that when building a custom Android-x86 installation, I discovered that [this alternate driver](https://github.com/rockchip-linux/kernel/tree/release-4.4/drivers/net/wireless/rockchip_wlan/rtl8723bs) is more stable and resolves the issues with disconnects and network timeouts.

Figured I'd mention this, if there were a possibility you could include it maybe as the default driver OR as an optional one to be selected.

Thanks!

clee604, 2022-04-03:
Hi @ouija , thanks for tagging my issue and discovering this alternate driver. I've only recently been learning Linux commands, are you able to provide some rough instructions on how to install this driver so I can test it as well? Thanks!

ouija, 2022-04-03:
It's a bit of a complex process, in that you need to compile the module (driver) _when_ building the kernel (for chromeos/brunch) OR build one for a similar kernel version and architecture.   

When I have time I'll attempt to do this and post back my results (and instructions on how to use this driver) provided I can get it working!

ouija, 2022-04-15:
I've managed to rebuild the alternate **rtl8723bs** module for Brunch/ChromeOS for all available kernel versions (click [here](https://www71.zippyshare.com/v/lvYlVkuM/file.html) to download)

This certainly seems to provide a more stable connection in brunch/chromeos for my device than the staging driver that is included.

To install, you'll need to download the zip file, and use the `8723bs.ko` that corresponds to the kernel you are booting with.

**PROCEED WITH CAUTION!**  I take zero responsibility if you mess anything up or if these instructions are unclear.

Download the file and open the .zip with File Explorer in ChromeOS so that it is mounted.

Press `CTRL-ALT-T` to open a crosh window, and then type `shell` followed by `sudo su -` to enter the shell as the _root_ user.

To remove/uninstall the current kernel module for the `rtl8723bs`, type `rmmod r8723bs`

Verify this module has been unloaded by running a `lsmod | grep 8723` command and see if there is anything listed.

**Please ensure that you use the same module compiled for kernel version as what you are currently running** _(if you don't know what kernel version you are running, you can type `uname -a` in the terminal shell and should see the version in the output)_

Now you can test the downloaded module to ensure it will load in your current enviroment, by typing:
`insmod /media/archive/rtl8723bs-alt.zip/rtl8723bs/<your_kernel_version/8723bs.ko`
If it loaded successfully, you shouldn't see any output, and if you do another `lsmod | grep 8723` you should see it listed, and if so, it will work successfully as a replacement module.

Unload the module before continuing, by running `rmmod 8723bs`

Now you copy the compiled kernel module from the mounted zip file to the kernel module directory and replace the existing one:
`cp /media/archive/rtl8723bs-alt.zip/rtl8723bs/<your_kernel_version/8723bs.ko /lib/modules/<your_kernel_version>-brunch-sebanc/kernel/drivers/staging/rtl8723bs/r8723bs.so`

And ensure that the file permissions are set to 644 by typing `chmod 644 /lib/modules/<your_kernel_version>-brunch-sebanc/kernel/drivers/staging/rtl8723bs/r8723bs.so`

And you can reboot, and hopefully if all went well, the module will be loaded on next boot and working better than previously.

Hopefully this is clear and makes sense (and works for whoever tries this);  I'm writing this off the top of my head, so again, I take _no responsibility_ if this breaks your brunch/chromeos install!

clee604, 2022-04-15:
Thanks!

The good news is it didn't break my install. The bad news is I still have an error when connecting, though now the error is "not associated" compared to unknown