sebanc/brunch #1499: How To Enable Secure Boot with Surface Pro and Brunch/ChromeOS

ouija, 2022-04-03:
Hey all,

Just wanted to share [how I managed to enable Secure Boot with Brunch/ChromeOS on a Surface Pro 4](https://github.com/ouija/SurfacePro4-Brunch-SecureBoot) using some info from another source I came across.

This method also incorporates the use of the rEFInd bootloader _(as opposed to using *Grub2Win*, which doesn't look as pretty or work with touchscreen devices)_, and also would require the `chromeos.img` to exist at the ROOT of one of your partitions (I have mine installed on a seperate partition from Windows formatted with EXT4)

I know that there is an alternate/better method for enabling Secure Boot with rEFInd, but this method was quicker and easier for me to achieve and, best of all, got rid of the ugly and annoying red banner / unlock icon that appears on a Surface Pro with secure boot disabled.

Note that with this EFI config, I've got the resolution set so that it actually causes the grub menu for Brunch/ChromeOS to be "hidden" _(yet still appear for 1s)_ so that I'm able to get into the options configuration menu by clicking the down arrow at the (blank) grub screen and get into it still.

Figured I'd share this here in case anyone else is looking to achieve the same thing!

clee604, 2022-04-04:
Hi,

Are you using rEFInd to chainload Grub? I have a different use case in that I want to be able to use the touchscreen to choose between booting Windows or Brunch. However I haven't been successful in figuring how to configure rEFInd to boot brunch. I currently use grub2win as per the current documentation, but like you said it doesn't support touchscreens so I have to connect a usb keyboard.

ouija, 2022-04-04:
Yes, using rEFInd with Grub2 that allows for secure boot.

Grub2Win requires a configuration file loaded from the Windows partition, but I'm using the basic Grub EFI with nothing more than the grub.cfg file that brunch creates when building the img _(with a timeout added)_

You can simply copy the `/EFI/grub` folder to your EFI parition and add an entry to your boot list _(using EasyUEFI if needed)_ and have it point to the `/EFI/grub/grubx64.efi`, and update the `/EFI/grub/grub.cfg` to match the .txt file generated when creating the chromeos.img via brunch

clee604, 2022-04-05:
Thanks for the instructions! I'm a bit wary of just copying files on the internet onto my EFI partition so I'm reproduce some of the configuration (also for my own learning). If I install rEFInd and Grub2, will rEFInd find the Grub2 bootloader automatically or do I need to add a Grub2 configuration entry to chainload into Grub2?

My ideal state is, boot into rEFInd by default where I can use the touchscreen to select 'Windows' or 'Grub2 (which would start Brunch)'

ouija, 2022-04-05:
rEFInd will find ANY `.efi` loader or file exists under the `/EFI` folder _unless_ otherwise specified in the `refind.conf` configuration file -- unless the `.efi` doesn't match the architecture of your system (32bit vs 64bit).

Qbit-Ten, 2022-07-14:
Hi, I'm trying to achieve the same thing with my SP4 but I can't find the EFI folder or the partition itself to replace them. Enabling secure boot will completely skip booting into brunch and throw a "No bootable operating system" error. How to access the EFI folder?

PatrickMitchellUI, 2022-07-17:
has anyone had luck enabling secure boot on an og surface pro