sebanc/brunch #1503: Problems with debug mode and ssh crosh

Brun0gameplay-12, 2022-04-07:
(translated text from Portuguese in google translator)

I installed chrome os 99 and brunch 100 on my computer and I am not able to open crosh even choosing debug mode when starting the system.

when I press "ctrl + alt + t" the crosh opens and closes by itself at the same instant.

so I found here you guys saying how to get it out of debug mode and that by default it already starts in debug mode, but mine not only is not opening directly in debug mode and I have to manually mark when I start the system as even entering debug mode o cross doesn't work.

with all these problems i installed chrome os and brunch 87 and it worked crosh, in it i marked pwa and then i updated brunch by pwa and then chrome os by pwa, but it did not solve the problem.

in addition I need to enter the chrome post developer channel I want to install steam without using linux and in the tutorial I saw you need crosh and the developer channel to install steam.

(tutorial on how to install steam that I found)

Enabling Steam on Chrome OS
On your supported Chromebook, switch to Dev channel. (This is Dev channel, not Dev Mode)
After updating, navigate to chrome://flags#borealis-enabled and set the value to
Enabled.
After restarting, open a crosh terminal with ctrl+alt+t.
Type “insert_coin volteer-JOlkth573FBLGa” and hit enter.
Follow the setup flow to install Steam.
Log in with your Steam account and start playing!

(end of tutorial)

I want to install steam post by it I can also run game maker studio 2

I had already tried to install it from the linux terminal of chrome os and it didn't open (I tested it with steam, the game maker itself, minecraft and soundrelay, none of these applications worked even with all the dependencies installed and the entire linux terminal updated), installed with all dependencies but it didn't open, so my hope is for steam that is being developed for chrome os

(translated text from Portuguese in google translator)

Brun0gameplay-12, 2022-04-07:
> porque você não é o único com o alt ctrl t problem state recovery, brunch e se você estiver usando legacy ou uefi e cpu

I couldn't understand what you said through the translator

PicPew00, 2022-04-07:
write what kind of cpu you have, what recovery image you are using and brunch

Brun0gameplay-12, 2022-04-07:
> escreva que tipo de CPU você tem, qual imagem de recuperação você está usando e brunch

Brunch r100 estável 20220402

Versão atual do Chrome OS - rammus R99

Intel core i3 2° gen - i3 2100