sebanc/brunch #1506: google play not work

jafer241177, 2022-04-11:
i use brunch 94 and crs rammus 96 when update to brunch100 and crs rammus 100 google play not work and tray power wash 
it work but when restart my pc it stopped again 
my processor intel Celeron 4020 " lenovo ideal pad flex 3 11.6
what i do i joined to group in telegram but no one solve 
please necessary find a solve because i cant update since brunch 94 and crs rammus 96 every time update it make same problem 
google play stopped and every app stopped 
thanks for time

DangJiYi, 2022-04-11:
"chrome://restart"

jafer241177, 2022-04-11:
same problem

darethehair, 2022-04-12:
This seems like the same problem reported in #1474, which I also have:

https://github.com/sebanc/brunch/issues/1474