sebanc/brunch #1507: How to set the size in MB ?

1457384613gh, 2022-04-11:
`sudo bash chromeos-install.sh -src chromeos_filename.bin -dst ~/tmpmount/chromeos.img -s size`
The size is in GB.
If the size I want to set is 61,036 MB=59.60547 GB , how could I input?

darethehair, 2022-04-12:
Not answering your question directly, but in my case I typically just want to 'fill' a partition, so rather than trying to figure out what size this is, I just try values that are too large/high.  The utility will tell you this, and then I just keep trying -- using 1gb smaller each time -- until it works/fits.

tejasraman, 2022-04-19:
Type the 59.xxxxx number

1457384613gh, 2022-04-19:
How about MB?

If GB，some number is too long to type, even is recurring decimal.

[image]

tejasraman, 2022-04-19:
I could try to modify the script…

tejasraman, 2022-04-19:
or not

Tejas Raman (he/him/his)

Contact me at ***@***.*** ***@***.***/0?redirect=mailto%3Atejasraman%40icloud.com&recipient=cmVwbHkrQVhEWUtEQ0laQzM0S1NPREhFMkRKUk9BTk5HSlRFVkJOSEhFTzc1NUZRQHJlcGx5LmdpdGh1Yi5jb20%3D)
My website: tejasraman.36bit.com ***@***.***/1?redirect=https%3A%2F%2Ftejasraman.36bit.com&recipient=cmVwbHkrQVhEWUtEQ0laQzM0S1NPREhFMkRKUk9BTk5HSlRFVkJOSEhFTzc1NUZRQHJlcGx5LmdpdGh1Yi5jb20%3D)

1457384613gh, 2022-04-19:
What?

tejasraman, 2022-04-19:
The script takes GB directly, so there is no workaround to this. Setting the partition size is the best way I can think of because about 1.2GB of space is taken by Brunch system files

Tejas Raman (he/him/his)

Contact me at ***@***.*** ***@***.***/0?redirect=mailto%3Atejasraman%40icloud.com&recipient=cmVwbHkrQVhEWUtER1VXSDVYSjdWSlBST0daTUdBTlJZSEJFVkJOSEhFTzc1NUZRQHJlcGx5LmdpdGh1Yi5jb20%3D)
My website: tejasraman.36bit.com ***@***.***/1?redirect=https%3A%2F%2Ftejasraman.36bit.com&recipient=cmVwbHkrQVhEWUtER1VXSDVYSjdWSlBST0daTUdBTlJZSEJFVkJOSEhFTzc1NUZRQHJlcGx5LmdpdGh1Yi5jb20%3D)

1457384613gh, 2022-04-19:
But the size of the partition is 128 GB≈122,070 MiB.
I want a chromeos.img of 64 GB≈61,034 MiB, and remain of 61,036 MiB≈64GB.

So I should `sudo bash … 61,034 MiB`

1457384613gh, 2022-04-20:
Just only MiB, could you?
Type the number without unit is of MiB.
If users type number with $GiB, it will ×1024 automatically.

$GiB
GiB='×1024'

`sudo bash … 64$GiB`=`sudo bash … 65536`

Could you?

tejasraman, 2022-04-20:
What is the remaining for? (is it for chromeOS?)

Tejas Raman (he/him/his)

Contact me at ***@***.*** ***@***.***/0?redirect=mailto%3Atejasraman%40icloud.com&recipient=cmVwbHkrQVhEWUtERUtHRlVKUVJQSTJLQ0pWTk9BTlI1U05FVkJOSEhFTzc1NUZRQHJlcGx5LmdpdGh1Yi5jb20%3D)
My website: tejasraman.36bit.com ***@***.***/1?redirect=https%3A%2F%2Ftejasraman.36bit.com&recipient=cmVwbHkrQVhEWUtERUtHRlVKUVJQSTJLQ0pWTk9BTlI1U05FVkJOSEhFTzc1NUZRQHJlcGx5LmdpdGh1Yi5jb20%3D)

1457384613gh, 2022-04-20:
What?
The link for you is following. https://github.com/sebanc/brunch/issues/1507

tejasraman, 2022-04-20:
What?
That was my stupid email sugnature

1457384613gh, 2022-04-20:
> What is the remaining for? (is it for chromeOS?)
> 
> 

For installing Android x86_64 with gearlock and MagiskSU.

Apks needing SU and several kinds of modules could be installed in it.

tejasraman, 2022-04-20:
I've used Magisk before, believe me.

Make 2 partitions. This is an answer.

1457384613gh, 2022-04-20:
I have another idea with inspiration of Android x86_64.

Dual boot

Don't re-size the img file when installing ChromeOS with brunch.

Just turn the bin file into img file and rename without re-sizing.

Make a new folder named `stage` following the x.img file and x-grub.txt file.

Edit grub and so on to mount the folder named `stage` in place of the last partition named `stage` into img disk.

If so, ChromeOS could make full use of all the space of the ext4 partition.

tejasraman, 2022-04-20:
No... .bin does not = .img. Brunch patches it to work too
You should try Bliss OS , not Android x86

1457384613gh, 2022-04-20:
OK，patch without re-sizing.

I want to mount a folder as stage out of img disk.

tejasraman, 2022-04-20:
The thing is, in Brunch dualboot, no system partitions are made - they are all _inside the IMG file_. It would be best to leave the

tejasraman, 2022-04-20:
The last version of android-x86 is 9.0. Bliss OS has android 11 and 12 already.

1457384613gh, 2022-04-20:
I'm sorry but how about `Downloads`?

ext4: 
/xxx/x.img
/xxx/x-grub.txt
/xxx/Downloads   (a folder)

Mount /xxx/Downloads as Downloads into img.

Or link Downloads to Downloads.

tejasraman, 2022-04-20:
The img contains all the partitions like the custom ROOTC and system reserved - you can't just create some weird custom IMG file with only the plain recovery image, as that needs files extracted and put in the right places.

1457384613gh, 2022-04-20:
OK, I get it.
How about f2fs?
It's faster friendlier to flash than ext4.
Could img disk be put into f2fs and then be booted into f2fs?

tejasraman, 2022-04-20:
You could copy the img to an f2fs partition (or try), as the grub entry has no partition specified. Do you have an add though? F2fs isn't much use if not

tejasraman, 2022-04-20:
Errors - silly autocorrect meddling with my comments, I meant ssd

tejasraman, 2022-04-20:
There's no guarantee it will work

1457384613gh, 2022-04-20:
Thank you very much for your early reply and how earnest you are!

tejasraman, 2022-04-20:
Thanks! (you said that in the colter r95-6-7 post too)

1457384613gh, 2022-04-20:
What do you thank me for?
Just a bug around NTFS.

I think that I can help you less than you can help me.

tejasraman, 2022-04-21:
Did it work?
Did you manage to get chrome is installed?

1457384613gh, 2022-04-21:
I haven't try.
I just have a 75GB chromeos.img into ext4 for stable use.

tejasraman, 2022-04-21:
Ok... you have approx. 53 GB for the Bliss OS/android-x86...

1457384613gh, 2022-04-21:
https://github.com/1457384613gh/rEFInd-theme-named-Yours
This is my repository.
Would you like to try?

[image]

tejasraman, 2022-04-21:
Ok... I love the concept of using a jpg file. Would this open in something other than office 2021?

tejasraman, 2022-04-21:
FYI I have WPS and Office365 for Mac

tejasraman, 2022-04-21:
How many OSes do you have?
I see opencore, fedora, brunch,

1457384613gh, 2022-04-22:
> FYI I have WPS and Office365 for Mac

Maybe，but I haven't try. I cannot promise the resolution of the banner exported is right. `Office 365` could be tried. 
I will make `BannerEditor.psd` for `Photoshop` to edit, because many users using Mac have `Adobe Zii`.

1457384613gh, 2022-04-22:
I haven't counted.

[image]

[image]

[image]

[image]

tejasraman, 2022-04-22:
I mean how many do you have installed.
Did you fix your chrome problem?

1457384613gh, 2022-04-22:
I‘m sorry but they are only icons.
All I have installed is a few.
[image]

Brunch works very well
[image]

tejasraman, 2022-04-22:
Great.., did android x86/bliss is work?

1457384613gh, 2022-04-22:
> Great... did android x86/bliss is work?

Yeah, they can work well, too.
If bliss, the folder containing `grubx64.efi` of bliss should be renamed `blissos` for its icon.

tejasraman, 2022-04-22:
Did they work for you? Remember the original issue?

1457384613gh, 2022-04-22:
> Did they work for you? Remember the original issue?

I haven't tried to type a decimal.

1457384613gh, 2022-04-22:
But I still think MiB is better than GiB, which called GB though.
[image]

tejasraman, 2022-04-22:
Don’t grow a decimal, type a whole number = to the size of the partition. Now, is the issue resolved?
I will try your refind theme at some point - my computer relies on refind as I have 4 OSes (Pop, Win11, Arch, ChromeOS) and I don’t like grub.

tejasraman, 2022-04-22:
GB = gigabyte
GiB = gibibyte

I
1 GB = 0.93 GiB

1457384613gh, 2022-04-22:
Ok, thank you for your coming experience.
And what I really want is make a chromeos.img with size of real GB not GiB, so I want to use MiB to turn GiB into around GB.
If `sudo bash ... 75`, I get a chromeos.img of 75GiB not 75GB.

tejasraman, 2022-04-22:
69 GiB = 75 GB (approx)

1457384613gh, 2022-04-22:
There is a new release. Adjust fonts.
[image]

tejasraman, 2022-04-22:
Try 69 gib

1457384613gh, 2022-04-22:
Select fonts whichever you would like.

Thank you for the number though the margin of error is more than 0.25%.

tejasraman, 2022-04-22:
Its ok to have so,e space

tejasraman, 2022-04-23:
That is the best I can do. It doesn’t accept decimals (I tried)

1457384613gh, 2022-04-23:
Considering the partition alignment, the size of partitions are multiple of 2MiB.

So, 2MiB is like a bridge between real GiB and almost GB.

Those who always use Windows like using GiB,
for whom, here is nothing;

Those who always use Mac or Fedora like using GB,
for whom, here is this.

- 2GB∧4=  16GB≈ 15,258 MiB
- 2GB∧5=  32GB≈ 30,518 MiB
- 2GB∧6=  64GB≈ 61,036 MiB
- 2GB∧7= 128GB≈122,070 MiB
- 2GB∧8= 256GB≈244,140 MiB
- 2GB∧9= 512GB≈488,282 MiB
- 2GB∧X=1024GB≈976,562 MiB

They could have gotten a `.img` with size of almost 2GB∧n, and n=4,5,6,7,8,9,10…

Maybe, it's obsessive-compulsive disorder.

However, `disk tool` of Mac and real disk capacity have that which make people have better feelings.

If `sudo bash … [number] #MiB`, we will get a `.img` disk with all kinds of sizes of both really GiB and almost GB.

1457384613gh, 2022-04-23:
Upside down.

If we want to get a `.img` disk of almost 64GB, we could have typed `sudo bash … 61,036 #MiB`.
🤣🤣🤣
So if only MiB!

tejasraman, 2022-04-23:
🤣🤣 😁 was that really  the solution? Typing #MiB after the number in bash?
That is just what I needed because my disk wasn’t big enough.
You helped me more than I helped you.

1457384613gh, 2022-04-23:
Look this is my disk.

[image]

1457384613gh, 2022-04-23:
Look, here is my SSD.
[image]

tejasraman, 2022-04-23:
Great!
Where’s the android?

1457384613gh, 2022-04-23:
Here is ext4, now. I have updated it, just now!
[image]

Android is here. I have updated again, just now.
[image]

[5 more replies]

closed 2022-05-17 by 1457384613gh