sebanc/brunch #1508: Trackpad very slow on HP AM080NA Notebook

DaTrafficCone, 2022-04-11:
Hi,
I recently installed brunch on my
HP Notebook and the trackpad has been acting up.
It is extremely slow and changing
the speed does nothing.

KarlS0808, 2022-04-13:
You can apply the alternetive trackpad config by going into brunch config

DaTrafficCone, 2022-04-13:
Thanks, ill try that

DaTrafficCone, 2022-04-29:
It didnt work, i tried everything, i even used a HP chromebook recovery image.

Mou786Aha, 2022-06-19:
> Hi, I recently installed brunch on my HP Notebook and the trackpad has been acting up. It is extremely slow and changing the speed does nothing.

Did you try sensitivity settings in settings? If not, find the setting related to trackpad settings and increase it.

DaTrafficCone, 2022-06-19:
> > Hi, I recently installed brunch on my HP Notebook and the trackpad has been acting up. It is extremely slow and changing the speed does nothing.
> 
> Did you try sensitivity settings in settings? If not, find the setting related to trackpad settings and increase it.

Yes, i tried all of the fixes and the trackpad is still really slow and choppy.

ewitsdevan, 2022-12-15:
I turned off "Enable tap-to-click" and it fixed my issue for me. Short and fast movements would register as taps.