sebanc/brunch #1509: How to open the brunch configuration menu

DangJiYi, 2022-04-12:
where is this "ChromeOS (settings)"?
[image]

Why is my interface like this
[image]

KarlS0808, 2022-04-14:
You can also go into it by doing ctrl+alt+f2 and logging in as root then doing edit-grub-config or edit-brunch-config

DangJiYi, 2022-04-14:
> You can also go into it by doing ctrl+alt+f2 and logging in as root then doing edit-grub-config or edit-brunch-config

No, my interface is still the old version. I don't know how to open the new configuration menu

tejasraman, 2022-04-15:
Update brunch using PWA. See #1512

closed 2026-01-21 by github-actions[bot]