sebanc/brunch #1512: How can I update the Zork recovery image version if I am already using ChromeOS in single boot

Anuragisme, 2022-04-14:


tejasraman, 2022-04-14:
Use the PWA

tejasraman, 2022-04-14:
There are instructions on the troubleshooting page but if you want but I can provide a video

Anuragisme, 2022-04-14:
> There are instructions on the troubleshooting page but if you want I can

Please provide the video.

tejasraman, 2022-04-14:
Ok..... i'll try to make my existing brunch install do the trick. 

I'll send a video on the installation and usage of the PWA as a way to update Brunch, but it is best (for me at least) to use the enable_updates brunch option and use settings to update. I will edit this comment and add the video in about ½ an hour.

tejasraman, 2022-04-15:
** UPDATE ** - Here is a video 

https://user-images.githubusercontent.com/96961804/163502788-cfe07d96-768c-4f71-be95-a15808a2b464.mp4

Anuragisme, 2022-04-15:
Thanks a lot !!! for helping me.

tejasraman, 2022-04-15:
Did it work?

Anuragisme, 2022-04-16:
Yes it worked. Now I am getting updates.

Anuragisme, 2022-06-24:
Could you send me that video again. I want enable updates on chrome os after reinstalling it. Please !!!

tejasraman, 2022-06-24:
Yeah, here is the video
> Chromebook.video.mp4

tejasraman, 2022-06-25:
Here:

https://user-images.githubusercontent.com/96961804/163502788-cfe07d96-768c-4f71-be95-a15808a2b464.mp4

Anuragisme, 2022-06-26:
Thank you very much.

tejasraman, 2022-06-27:
Anytime!

tejasraman, 2022-06-28:
I can’t really get any info. In ChromeOS (settings), enable verbose boot when asked. Take a slow-motion video of the the onscreen text.

Also, what is our laptop model?

Anuragisme, 2022-06-28:
But How to get into setting. Chrome OS is not starting.

Laptop Model : Lenovo V15 ADA
Processor : AMD Athalon Silver 3050U

Anuragisme, 2022-06-29:
> In the menu, select ChromeOS (Settings) underneath "ChromeOS" with the
> down arrow and press "ENTER", select the normal options and type "YES" at
> verbose boot.
>
> I can make another video if needed ;)

Yes, I have done that. What's the next step? I am sharing the video with
you.

https://photos.app.goo.gl/Nmm1tLCGBh8Y5L8V8

Anuragisme, 2022-06-29:
This problem had occurred twice when I updated Chrome OS. and when i
reinstalled Chrome OS and updated it then happened the same thing.

tejasraman, 2022-06-29:
2 things:
- Try kernel 5.10
- Try `pci=noafb` as a boot arg in cmdline boot args

tejasraman, 2022-06-29:
ohhhhh, try updating brunch first to the **stable release**

Anuragisme, 2022-06-29:
Yes I have tried updating Chrome OS after updating brunch to 102

tejasraman, 2022-06-29:
pci=noafb? Kernel 5.15?
These are some fixes that might help you

Anuragisme, 2022-06-29:
I tried them but not working. 😞😞😞

Anuragisme, 2022-06-30:
Yes I tried all  that you told me. Yes I am on Zork

tejasraman, 2022-06-30:
Try using the chromeos 101 recovery image with brunch 102 please? Send me a verbose after that pls?

Anuragisme, 2022-06-30:
Sure, 'll send soon.

Anuragisme, 2022-06-30:
Is there any way to do this other than reinstalling Chrome OS.

tejasraman, 2022-06-30:
thanks for being prompt about that...
If you want, I'll run a test with brunch 102 on my end... 
Before that, try "nomodeset" as a kerel option

Anuragisme, 2022-06-30:
Ok

tejasraman, 2022-06-30:
Thanks for responding in <30 seconds! ;)

Anuragisme, 2022-06-30:
😆😆😆

Anuragisme, 2022-06-30:
Thanks you so much for you assistance.

tejasraman, 2022-06-30:
did it work?

Anuragisme, 2022-06-30:
Doing

Anuragisme, 2022-06-30:
Please wait

Anuragisme, 2022-06-30:
No it didn't work.

tejasraman, 2022-06-30:
I've reached the end of my abilities. @sebanc do you know?

Anuragisme, 2022-06-30:
No

tejasraman, 2022-06-30:
what?

Anuragisme, 2022-06-30:
I don't know about @sebanc

tejasraman, 2022-06-30:
He is the main brunch developer

Anuragisme, 2022-06-30:
How can I contact him ?

tejasraman, 2022-06-30:
the @ mention sends a notification to his feed. He might look at the issue? Just wait a bit.

Anuragisme, 2022-06-30:
Ok

Anuragisme, 2022-06-30:
What are you doing.

tejasraman, 2022-10-11:
In the menu, select ChromeOS (Settings) underneath "ChromeOS" with the down arrow and press "ENTER", select the normal options and type "YES" at verbose boot.

I can make another video if needed ;)