sebanc/brunch #1518: linux is missing

Pegasus20x, 2022-04-15:
i have brother DCP-T300 printer ,but it seems like it's not supported on chrome os. i have the linux driver for the printer. will it fix the issue?
the problem is i can't find linux installer anymore. i tried powerwash but still missing

tejasraman, 2022-04-15:
Enable virtualization in BIOS
What recovery image did you use?

tejasraman, 2022-04-15:
Ohhh. I say your other post - you have a 4th gen I7.
Use Rammus image 100 with Brunch 100

Pegasus20x, 2022-04-15:
> Enable virtualization in BIOS What recovery image did you use?

It's enabled

Pegasus20x, 2022-04-15:
> Ohhh. I say your other post - you have a 4th gen I7. Use Rammus image 100 with Brunch 100

I use rammus 100 with brunch 100

tejasraman, 2022-04-15:
Ok... I really have no idea about this. Take a screenshot of settings pls?

Pegasus20x, 2022-04-15:
> Ok... I really have no idea about this. Take a screenshot of settings pls?
[image]

tejasraman, 2022-04-15:
Oh... thanks for the screenshot! Did you check advanced - developers in settings?

Is this a managed account? I had a managed account and it didn't support Linux - it is an option admins can enable and in Family Link(if that is what you are using) Linux cannot be enabled.

Also(have a **talk** with your admin if it is managed - that is the problem) is this in an NTFS partition? (if so, 😢) and switch to EXT4 with a Linux live USB (if you don't have Linux download a Linux Mint ISO and see [here](https://trendoceans.com/how-to-create-a-bootable-usb-using-rufus-for-linux-distributions/) to make a USB)

Pegasus20x, 2022-04-15:
> Oh... thanks for the screenshot! Did you check advanced - developers in settings?
> 
> Is this a managed account? I had a managed account and it didn't support Linux - it is an option admins can enable and in Family Link(if that is what you are using) Linux cannot be enabled.
> 
> Also(have a **talk** with your admin if it is managed - that is the problem) is this in an NTFS partition? (if so, 😢) and switch to EXT4 with a Linux live USB (if you don't have Linux download a Linux Mint ISO and see [here](https://trendoceans.com/how-to-create-a-bootable-usb-using-rufus-for-linux-distributions/) to make a USB)
> 
> Also, I notice you enabled the light/dark theme flag :)

i am the admin of my account . i installed chrome os last year and i had the linux . i do have the partition in NTFS.

tejasraman, 2022-04-16:
> Also(have a talk with your admin if it is managed - that is the problem) is this in an NTFS partition? (if so, 😢) and switch to EXT4 with a Linux live USB (if you don't have Linux download a Linux Mint ISO and see here ***@***.***/0?redirect=https%3A%2F%2Ftrendoceans.com%2Fhow-to-create-a-bootable-usb-using-rufus-for-linux-distributions%2F&recipient=cmVwbHkrQVhEWUtESDRYQTdURlhOQjU1VVk2TVdBTTRTN0RFVkJOSEhFUFdYVlVRQHJlcGx5LmdpdGh1Yi5jb20%3D) to make a USB)

weird... If this started recently, I cannot help you past this. I have seen a few problems with people using NTFS, so maybe put it onto ETXT4?

tejasraman, 2022-04-16:
and you will have to print from linux for the driver to work.

tejasraman, 2022-04-16:
@sebanc is Brunch 100 broken?

Pegasus20x, 2022-04-16:
> and you will have to print from linux for the driver to work.

so you're saying that even if i managed to install linux and the linux driver , i still won't be able to print using chrome ,and i'll have to print using linux terminal?

tejasraman, 2022-04-17:
No… You will have to download a Linux version of Chrome or Edge or Firefox and print from it
But you cannot print from the built in chrome unless you use crosh and types shell to access a system level Linux shell (there you can install the driver as system).