sebanc/brunch #1528: Google Play Terms of Service error

MineForce899, 2022-04-24:
This bug is on Rammus 94 and Rammus 100
I see ```Google Play Terms of Service cannot be loaded. Please check your network connection and retry``` after I log in to my google account on first startup. If I reboot my computer it will show me normal login screen but if I log in my account it will show this error again. If I login as a guest network is working fine and everything except google play too.
Crosh is working in a guest profile too, so is there any way to download it manually and put it somewhere?

MineForce899, 2022-04-24:
So it somehow got fixed but Google Play is still not working - it tells me this error but in window (like installing linux subsystem)

Y0seph, 2022-05-14:
Same issue here
Using a Chromebook Plus v2...

KushalSM, 2022-05-15:
Yeah same issue with hatch 100 too

mrwhistles, 2022-05-17:
The latest brunch release doesn't advertise 101 support but I enabled updates, updated to 101 using a guest session and that seemed to solve the problem.

KushalSM, 2022-05-17:
I don't think that is the issue, what I feel is like, I kept switching from board to board to find the one I found perfect for my computer, and at the point when I found it, it didn't let me in, it's mostly a network issue, or the reason that you keep logging into many times like I did, what worked for me before was changing my password, then using my phone to sign it out of all the chrome os logins, and it worked.

Another thing you can try is do the same thing above, and also add a 2-step auth, and I was able to login again succesfully. Some say it could be a google side server issue too, so maybe give it some time for that in case that is the reason.