sebanc/brunch #1535: Android 11

KushalSM, 2022-05-11:
How to get android 11 on brunch chrome os, I was not able to find any solution on the internet, and hence came here.
I have got a Intel i5 8th Gen, 8gb ram, and an Intel uhd 620 graphics card, pls help

KushalSM, 2022-05-11:
And if there is a solution, pls document this in your wiki too

fenginsc, 2022-05-12:
I'm also looking for a way to upgrade the Android version, but It seems that the Android version of chrome OS has always been 9, and there is no version 11 as you said.

midozalouk, 2022-05-13:
it's related to the recovery 
i am using hatch recovery and it has android 11

KushalSM, 2022-05-13:
> I'm also looking for a way to upgrade the Android version, but It seems that the Android version of chrome OS has always been 9, and there is no version 11 as you said.

Yes that is true, the only source I found is that android 11 is going to come in so and so chromebooks, but since this isn't a chromebook, but one of the closed issues showed him getting android 11 on his

KushalSM, 2022-05-13:
Yes, the same was found from another user in the closed issues. I also found that too from a reddit page, but I also remember reading somewhere that rammus has got it too

KushalSM, 2022-05-20:
Well, it's concluded that the android 11 arcvm comes to you based on your board, so will be closing this issue!

Thank you everyone here for taking your time and helping me with my issue! :)

closed 2022-05-20 by KushalSM