sebanc/brunch #1539: MacBook Pro 2011, Dual-Boot With Fedora, Not Boot

daudix-UFO, 2022-05-14:
Followed dual-boot tutorial provided here, updated grub, with fedora analog of grub-update, "grub2-mkconfig -o /boot/efi/EFI/fedora/grub.cfg", Don't boot u
[image]
[image]
p

1457384613gh, 2022-05-16:
Hi, I am a user of Brunch, too.
I have happened to see your issue.
Here my repository is, https://github.com/1457384613gh/rEFInd-theme-named-Yours
You can copy `img_file(loop,12): /efi/boot` into `ESP: /EFI` and rename it `brunch`.
And then edit `grub.cfg`.
[image]

1457384613gh, 2022-05-16:
You can download [EFI.zip](https://github.com/sebanc/brunch/files/8696850/EFI.zip).
[image]

closed 2022-05-16 by daudix-UFO

Dakai, 2022-06-19:
Hi, @daudix-UFO, I have the same Linux setup as you and the same error output when installing dual boot, did you find a way to fix it? Thanks.