sebanc/brunch #1544: Unable to start brunch

bybenjamin, 2022-05-19:
I have installed chrome os with brunch on another computer many times before. Fydeos works fine.
I did a single install without any problems. But brunch doesn't open.
Recovery: lantis 100 (I also tried rammus)
Brunch version: Brunch 100
Device features
CPU: N4500
Graphics Card: Intel UHD
Ram: 4GB
I am sharing the verbose output. Help me @sebanc 
[](https://photos.app.goo.gl/UKrEfuncEgWYDypP6)

sebanc, 2022-05-21:
"ui main process termninated with status 2" error message usually occurs when chromeos cannot access the kernel modules, it is generally due to:
- An incorrect install
- Booting a brunch hdd install with a brunch usb plugged in
- 2 brunch dual boot installs at the same path but in different partitions

Sorry but I highly doubt this issue is a brunch one.

bybenjamin, 2022-05-21:
I've installed many times before and all went smoothly.  I'll try again and let you know the result.

bybenjamin, 2022-05-23:
There is no problem in hdd and usb partitions.
I leave screenshots.
Boot problem still persists. what is the problem @sebanc 
[image]
[image]