sebanc/brunch #1548: No hardware decoding, game streaming

Rattonikus, 2022-05-22:
When using streaming apps, such as steamlink, parsec, and moonlight, they all seem to fail because of no hardware decoding, steamlink crashes, parsec gives a decode_init error, and moonlight fails to connect, any information on this would be appreciated, thank you. Im using an Acer Aspire N15W5, Intel Celeron N3050, 2GB ram, integrated graphics.

Rafostar, 2022-08-17:
I also encountered this recently on Intel Atom z3795 (Bay Trail) with "rammus" recovery image. Using "coral" (Acer Chromebook 11 C732) instead fixes this issue completely for me. Although I had to additionally install `h264ify` extension to force h264 instead of AV1 which isn't supported by my hardware. Tested with YouTube and Stadia. Both work perfectly fine with "coral" image. Also Android subsystem works much faster for me in it then it did with "rammus".

blintwave, 2022-08-17:
> I also encountered this recently on Intel Atom z3795 (Bay Trail) with "rammus" recovery image. Using "coral" (Acer Chromebook 11 C732) instead fixes this issue completely for me. Although I had to [...]

What brunch and chrome version are you using right moment?

Rafostar, 2022-08-17:
> What brunch and chrome version are you using right moment?

Latest (r103). Kernel 5.10. I tried first with "rammus" as wiki suggested, but its meant for newer core Y based CPUs and HW accel does not work in it. So I tried "coral" which is taken from Celeron N3350 notebook (Atom based cores). This one worked without any problems. All installed on USB pendrive for experimenting yesterday. I will probably keep testing it for next few days and install permanently to eMMC once r104 is out (I am waiting for Dark Theme support).

blintwave, 2022-08-17:
> Latest (r103). Kernel 5.10. I tried first with "rammus" as wiki suggested, but its meant for newer core Y based CPUs and HW accel does not work in it. So I tried "coral" which is taken from Celeron [...]

I dunno what happend with rammus tho, im the past it worked for me but from release 98 or so i did not have hw acceleration anymore. I even tried another recovery from HP I think but that did not work either. Although running Chrome Flex everything works.

1 monty ago I also made an issue rapport...

https://github.com/sebanc/brunch/issues/1578

But till now did not investigated further, but I will try you brunch and recovery combo and see how it goes. Probably the weekend when I have more time.

blintwave, 2022-08-19:
> Latest (r103). Kernel 5.10. I tried first with "rammus" as wiki suggested, but its meant for newer core Y based CPUs and HW accel does not work in it. So I tried "coral" which is taken from Celeron [...]

Alright, just a headsup. I installed the coral image like you suggested and now I have HW acceleration again. I will post this as well on my thread too. Thanks for the tip!

3bagorion33, 2024-05-07:
> I also encountered this recently on Intel Atom z3795 (Bay Trail) with "rammus" recovery image. Using "coral" (Acer Chromebook 11 C732) instead fixes this issue completely for me. Although I had to [...]

You made my day, man! Rammus wouldn't load at all because of a video driver error, and Coral showed the welcome screen. I'm going to install on my internal disk now and continue.

closed 2026-01-19 by github-actions[bot]