sebanc/brunch #1549: Not getting internet in chrome os 101

ganeshambalakkara, 2022-05-23:
Updated to Chrome os 101, wifi is showing connected but not getting internet, tried ethernet also. Help to fix this.

Jforte20, 2022-05-25:
I am having the exact same issue. Anyone have an idea on how to fix it?

gtaws, 2022-05-25:
Does brunch support 101 yet? I thought author said mixing versions of not a good idea.

ganeshambalakkara, 2022-05-27:
Powerwash and revert

ganeshambalakkara, 2022-05-27:
U can rool bavk

Firnin, 2022-06-07:
Same issue, I am using Acer C720P (rammus recovery). Yesterday I upgraded to the latest versions of Brunch and ChromeOS (was previously on v97), did a Powerwash since I had a Blackscreen after logging in. Setting up the Chromebook was impossible though, so I reverted back to Chrome OS 100, and everything worked fine (except setting up Linux)

allancoding, 2022-06-09:
Same here ChromeOS 101 recovery image and 101 brunch framework did work the wifi connected and it made requests to the  internet (I checked with pihole). The only way for it to work was with guest mode because you an sign in with no connection to the internet. But if you checked diagnostics it said something like HTTP can't connect through firewall. I went back to version 100 and it worked.

zbieg, 2022-06-10:
I found the following solution on reddit (shell command):
sudo /etc/init.d/networking restart
Let me know if it helps.
Source: https://www.reddit.com/r/chromeos/comments/v5y5u5/psa_crostini_cannot_connect_to_the_internet/

zakiloki0, 2022-06-17:
> I found the following solution on reddit (shell command): sudo /etc/init.d/networking restart Let me know if it helps. Source: https://www.reddit.com/r/chromeos/comments/v5y5u5/psa_crostini_cannot_connect_to_the_internet/

Not work, to me show "command not found"

In the /etc/init.d location only is functions.sh file.

any news about this problem with the internet conection in 101 @sebanc? Thanks

JohnnyLombax, 2022-06-21:
I've been having the same issue as zakiloki0. Tried the suggested fix posted on reddit to no avail.  Tried some different kernels and Wi-Fi options from the config without any progress as well. I'm running on an Inspiron 20 3043 using ChromeOS 101 on Brunch r101 stable 20220529.

Tried running ChromeOS 100 with the older build, but was unable to get it to boot for some reason. Hangs on the patching screen before ChromeOS launch for hours.

graphixillusion, 2022-06-26:
Same thing is happening with build 102

zbieg, 2022-06-27:
I think there is something wrong with the DNS settings/configuration .

In fact, the internet connection works. The only problem is with the correct interpretation of the domain names of the websites/URLs. When I login as a guest and enter the IP address instead of the domain name, the websites loads correctly (e.g. http://142.251.40.36 or http://185.151.30.155 or http://80.48.169.1 or http://198.35.26.96 etc.).

Maybe we just need to copy some DNS related configuration files from the older ChromeOS build?

graphixillusion, 2022-06-28:
@zbieg if you try to ping some domain in the shell, the DNS resolution of the selected domain works. The problem is in the GUI section of the OS. It wouldbe interesting to see why this is happening.

PicPew00, 2022-08-07:
any solution??

Alex2772, 2022-08-11:
> any solution??

Use the version 100. Worked for me.

Lukazitalo, 2022-08-14:
Same here, in the version 101, 102 and 103, in Samsung RV420... :/

rayep, 2022-08-28:
Same issue for me too.

I used a fresh install of r103 on my Lenovo B490. Wifi connected sucessfully, but the "please wait" screen loops forever as I can see the error "Oops, network error" message flashes in between.

Browse as guest takes me to the the guest instance of chrome and no webpage loads, not able to open Cros Shell - it closes immediately.

Weird part is I can get to the developer session by pressing Ctrl Alt F2 and there Internet is working just fine, I'm able to ping, use curl to download files, etc. So something is wrong only during the initial setup process.

@sebanc any thoughts? Thank you!!!

Jejebond, 2022-10-22:
Same issue here. Wifi and Ethernet won't work on Chrome OS 101+ (106 tested). 

Using Brunch R103. My computeur is a Lenovo X220 (Intel 2nd gen).
Also, I have upgraded to AX210 wifi chip, Kernel 5.15

SimurgAnka, 2022-10-22:
Same issue

KawaiiGirl2001, 2022-11-23:
#1715

KawaiiGirl2001, 2022-11-23:
#1715
I have the same problem

SalmanFardeen, 2022-11-27:
Same problem 😪😪

michalss, 2023-02-11:
same here Dell 64xx

realPilotfish, 2023-02-15:
I had the same issue and some experimenting.
I installed proxies in my net (socks and squid). 
DNS ist not the problem (at least for me). 
I can open websites that use http using their IP, however I cannot open sites that use SSL, not by name and not by IP, connection hangs.
So I think this has something to do with SSL.

flashresaka, 2023-02-23:
Hello, I have the same problem with version 109, I have noticed that the time is not updated even if I enter as a guest to change it, the problem may be related

Model f200ca asus

tejasraman, 2023-03-04:
This has to do with missing instruction sets..... read the dmesg

amasnur, 2023-04-13:
No updates? I'm facing the same issue on latest Brunch and Chrome OS 111.

tejasraman, 2023-04-13:
No updates (and probably never)

madhits, 2023-05-19:
same issue for me on my hp touchbook that is very similar to samus, pixel book 2015. Any hope, Im using ramus 107.

kocoman2, 2023-08-21:
how to dl the 100 version for https://cros.tech/device/shyvana/, only shows 105 and up
thx

tejasraman, 2023-08-21:
> how to dl the 100 version for https://cros.tech/device/shyvana/, only shows 105 and up
> 
> thx
> 
> 

Here's a direct link to do version 100 from Google's servers: [https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_14526.69.0_rammus_recovery_stable-channel_mp-v2.bin.zip](https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_14526.69.0_rammus_recovery_stable-channel_mp-v2.bin.zip)

Also this is support for brunch. I'm assuming you want this for an actual chromebook? This isn't the right place to ask 🙂

tejasraman, 2023-08-21:
> same issue for me on my hp touchbook that is very similar to samus, pixel book 2015. Any hope, Im using ramus 107.
> 
> 
> 
> See: https://github.com/sebanc/brunch/issues/168
> 
> 
> 
> This is a swtpm or TPM issue and it appears since our machines use a older TPM or have it disabled, versions of chromeOS past 100 wont boot. There has to be a fix. 
> 
> 
> 
> UPDATE: There is no fix, google apparently made huge changes to something in there code with 101 that Brunch can not overcome (e-waste prevails again, RIP machines from 2013-2016). So Your best bet is Brunch 102 and ramus 100.

Actually haswell and newer work. It's actually related to cpu instructions (avx2) that those older chips do not support.

Timu5, 2023-09-03:
What about patching chrome executable? dmesg clearly says there the invalid opcodes are located.

tejasraman, 2023-09-03:
> What about patching chrome executable? dmesg clearly says there the invalid opcodes are located.

It’s not just chrome, it’s many of the chromeOS components too. The effort needed to binary patch chromeOS executables is just too much to be practical. It’s easier to restrict support to Haswell (which is already 10 years old fyi).