sebanc/brunch #1550: Linux error installing

channelup69, 2022-05-23:
Brunch 100 and Chrome stable build 101.

I have tried various solutions proposed and have not had any success. 
Tried: DNS
          DLC
          
Still no success

Tash254-ke, 2022-05-24:
I have this issue too. At first after the update, terminal was not opening. Couldn't even access my files. Removed linux and tried installing again and now I am getting an error when installing.

Tash254-ke, 2022-05-26:
> I tried the enable updates in Brunch config and it worked. Thanks

Had you enabled updates before?

channelup69, 2022-05-27:
That was the first time I had enabled updates.

Tash254-ke, 2022-06-02:
Just updated to brunch 101 and chrome OS 102 and linux installed

srang992, 2022-06-25:
**Tash254-ke's** solution works for me.

block6791, 2022-07-04:
Had the same issue with Linux installation under Brunch 102 and Brunch r102 stable 20220625. I enabled Chrome OS upgrades and let it upgrade to Chrome 102. However this resulted in a boot loop so I could not verify if Linux works under Chrome OS 102.

Tash254-ke, 2022-07-05:
> Had the same issue with Linux installation under Brunch r102 stable 20220625 and Chrome OS 101. I upgraded to Chrome OS 102 via the PWA Brunch. However I am still unable to install Linux. I also upgraded to Chrome OS 103 by enabling updates. However this resulted in a boot loop that caused me to reinstall Brunch and ChromeOS.

I am back here again. I accidentally update to 103 and had to reinstall... Now Linux won't install. How did you upgrade to 102 via brunch?

block6791, 2022-07-05:
> I am back here again. I accidentally update to 103 and had to reinstall... Now Linux won't install. How did you upgrade to 102 via brunch?

I used the Brunch PWA for this. When you start the Brunch PWA you will see the 102 update under Chrome OS updates. However Linux also failed to install under Chrome OS 102. I did another upgrade to 103 (via the "Enable Updates" option) but I got the same result: A bootloop.

Tash254-ke, 2022-07-05:
> I used the Brunch PWA for this. When you start the Brunch PWA you will see the 102 update under Chrome OS updates. However Linux also failed to install under Chrome OS 102. I did another upgrade to [...]

"Warning: Don't update your brunch install to ChromeOS r103 as it will result in a bootloop.
This issue is related to a recent change in the cryptohome package and notably to the lack of official TPM.
Brunch uses swtpm but I am not very familiar with it so this issue might take a while to be solved."

Sebanc sent out this announcement over on discord.

tecstudyap, 2022-09-19:
how to enable update ?