sebanc/brunch #1551: Fronzen on ChromeOS logo

marcoriesco, 2022-05-25:
My question is as follows, I'm using rammus 101, could I be using the wrong image? 
I'll pass my hardware, tell me the image with more compatibility please

Intel(R) Core(TM) i7-9700F CPU @ 3.00GHz
Radeon RX 580
MB Gigabyte Z390

I waited more than 15 minutes, does it take longer?

darethehair, 2022-05-25:
Probably the same as what I reported:

https://github.com/sebanc/brunch/issues/1546

kahunama, 2022-05-26:
> 
brunch R93 zork  INTEL I5-10400   AMD RX550  2g  Successfully started and started the android subsystem。brunch r100 zork same problem as you!

kahunama, 2022-05-26:
brunch R93 zork INTEL I5-10400 AMD RX550 2g Successfully started and started the android subsystem, brunchr100 zork crosh [ERROR:src/main  and  The android subsystem cannot be started!

sebanc, 2022-05-26:
Just to clarify, even though those 3 issues have the same consequences (stuck on the ChromeOS logo) I am pretty sure the root cause is different in all cases:
- i7-9700F has not iGPU so the only recovery which could work with the AMD gpu would be zork but this configuration is not supported as it is known to have a lot of bugs.
- Issue #1546, standard intel GPU, it is a specific issue with ChromeOS r101 that I have not identified and which seems to affect only some chromebooks.
- I5-10400 AMD RX550 this intel CPU has a iGPU, most likely we are talking about a desktop which has the display connected to the dGPU whereas to use rammus image the display would have to be connected to the iGPU.

tejasraman, 2022-05-27:
If you see what sebanc said, he mentions the iHD driver was used in replacement for i965. (I965 is still the default in Linux distros)

iHD supports Broadwell and up (it seems according to their website)

Oh... does this mean the end of Brunch support for older GPUs and Intel/AMD mixed configurations?

P.S - is Braswell compatible with iHD?

tejasraman, 2022-05-28:
i5-10400 - use volteer, not rammus for this PC

closed 2026-01-19 by github-actions[bot]