sebanc/brunch #1556: System restarts on ChomeOS logo

allancoding, 2022-06-08:
The error I can see is like ext4-fs on virtualization.
I enabled verbose mode. The video here ended to soon but it showed ChomeOS then system restart then grub again.
https://drive.google.com/file/d/1QObXAlNQ1OTP5J21-YWJYJ2c0B3u2LBh/view?usp=drivesdk

I do not know much about brunch. I do not know the errors. Please help.

sebanc, 2022-06-08:
Thanks for the video.

Try to re-create the image, there is an error regarding the data partition which is most likely the cause of your issue.

allancoding, 2022-06-08:
What size should I do? I have a 89.9 Gib partition.

allancoding, 2022-06-08:
How much space should be available because it says 1.1 Gib available? And I made the image 77 Gib.

allancoding, 2022-06-08:
I think the boot problem was with using Kernel 5.15. 
Then I used Kernel 5.4:
I can access local ip addresses. I checked the log on my router and it is making the request and returning them.

closed 2026-01-19 by github-actions[bot]