sebanc/brunch #1557: microphone not working on xps 13 9310

6itH, 2022-06-08:
I'm running version 101 with kernel 5.15. Everything works perfectly but the microphone. I tried all the options but nothing worked. On Chrome OS Flex everything works fine (microphone included) with kernel 5.10

Any ideas? Thanks

librehat, 2022-06-10:
just found out microphone not working with my Samsung Galaxy Book Pro 360 as well (also Kernel 5.15). Tried both internl_mic_fix and internal_mic_fix2, none of them works.

channelup69, 2022-06-10:
Im using a Lenovo x1 carbon 5.10 kernal i7 11 gen and my mic works if I plug in a headset. In the audio settings i can toggle internal mic and head phone mic and both work along with the speaker toggle. Once I unplug from the head phones mic does not work. I was wondering if it was a driver issue. I had posted in another thread about SOF drivers. https://github.com/thesofproject/sof-bin

6itH, 2022-06-10:
Very similar specs with the xps. On my device I don't see an internal mic (but a dmic), so the internal_mic_fix wouldn't work I guess
[image]

channelup69, 2022-06-10:
[image]

channelup69, 2022-06-10:
Yes

6itH, 2022-06-29:
This has been fixed with release 102. Thank you

closed 2022-06-29 by 6itH