sebanc/brunch #1559: need to long press power button to wake up device (HP Elite x2 G3)

yeahman45, 2022-06-16:
I have installed brunch on HP Elite x2 G3 (specs here https://www.notebookcheck.net/HP-Elite-x2-1013-G3-i5-8350U-SSD-3k-2-in-1-Review.336734.0.html).

It works great except for waking up the device.

Once it sleeps, I need to long press the power button for 5-6 seconds for it to wake up and it takes 1 second for the screen to turn on.

The issue is sometimes if I press the power button for too long, the device restarts. (force reboot)

Is there a way to wake up the device by just single press the power button or any other keys like volume button?

Right now, if I single press the power button, the keyboard lights up (and turns off after a few seconds) but the device does not wake up.

On windows 10, it works fine; a single press turns the device on instantly. (from a deep sleep; battery drain is pretty much inexistent)

yeahman45, 2022-06-17:
anything I could try? what do I need to test and post to help make things advance?

tejasraman, 2022-06-17:
Did you enable s3 suspend in brunch options?

yeahman45, 2022-06-18:
> Did you enable s3 suspend in brunch options? That may be why.
> 
> Are there any sleep modes in the BIOS (like S3, S0/Modern Standby, S1)? Try other modes if it is set to S0 or Modern Standby
> 
> Try S3 and kernel 5.15 on brunch. This is the best I can do until I get more info so.....

I have enabled suspend S3 and it seems to work but I need to test it longer.
Thx man!
Out of curiosity, why is it not enabled by default? Does it consume more battery?

Now I need to try solve bluetooth which does not seem to work.

tejasraman, 2022-06-18:
S3 is a different power state where almost all system components except the RAM are powered off. Maybe Brunch was using hibernate as an alternative to S0 (what Win10 uses)?

FYI, S3 uses less power than S0

yeahman45, 2022-06-18:
thx for the explanation. seems to work well so far. my laptop does wake up when power button pressed or kyeboard button pressed or lid is open although there's some delay but I can live with that.

for bluetooth, no devices can be found via the UI but in crosh, this works : https://github.com/sebanc/brunch/issues/585

tejasraman, 2022-06-18:
This is because windows was using S0 (which has less delay because it is a higher power state than S3) and Brunch was using hibernate (which is the lowest power state your pc can use) - it shouldn't hibernate by default ;) it doesn't on my older, secondary laptop (a repurposed Samsung Chromebook 3 running Brunch)p because it's past the AUE)

closed 2026-01-19 by github-actions[bot]