sebanc/brunch #1560: Remove Brunch bootsplash during startup

isti2415, 2022-06-18:
I am using zork 101 and i want to remove the brunch bootsplash from startup. How do i do that? I have looked at some old solutions and seems like "sudo edit-grub-config" has been updated and it doesnt take me to the grub.cfg file anymore. It just gives me a list of options now instead. Anyone can help me on this?

Mou786Aha, 2022-06-19:
> I am using zork 101 and i want to remove the brunch bootsplash from startup. How do i do that? I have looked at some old solutions and seems like "sudo edit-grub-config" has been updated and it [...]

But it provides the option at the last screen. You could use it there. You have to press 'enter' for every screen you don't want to make a change. In the last section it displays about various options related to the brunch logo.  WARNING: Do at your own risk. In my case changing any setting like this caused me 'Not booting up problem'. I don't know for others but this is the method actually

isti2415, 2022-06-19:
> 

actually i meant the first screen where it gives me the option to enter chromeos or chromeos settings

primerlee, 2022-07-12:
> 

how to change the time, i just want to boot it faster

Drakon1505, 2025-01-21:
@primerlee Mount the sdx12 partition as /tmp/p12 by executing the following commands:

`mount -o remount rw /`
`mkdir /tmp/p12`
`mount /dev/sdx12 /tmp/p12`

Step 3: Edit the Configuration
Use the following command to open the file:

`sudo nano /tmp/p12/efi/boot/grub.cfg`

Scroll down to locate set timeout and modify it as shown below:

`set timeout=0`

Save the changes by pressing ctrl + x and then typing y.

Step 4: Reboot the System
Execute the following commands to unmount the partition and restart the system:

Terminal window
`umount /tmp/p12`
`reboot`