sebanc/brunch #1563: Who's using this chromebook loop.

edschofield15, 2022-06-19:
I'm trying to install with the Rammus image on an Acer C740. I can get as far as the "Who's using this Chromebook" screen. But when I press next, it goes to the "please wait" screen and sticks there for a while (Intermittently flickering back to the wifi selection screen) and eventually goes back to the "Who's using this Chromebook".

I can get into "browse as guest". But when I try to load any website, the pinwheel spins continually but nothing happens (Doesn't seem to load or time out). 
If I go into the terminal, I can ping 8.8.8.8 and if I do "curl ifconfig.io" it returns the correct address. Also when I run the connectivity diagnostic, it passes as far as the last option, then says "can't get through firewall to https websites". 

I've also tried with a USB ethernet adapter. Which is picked up and seems to be working, but still does the same as WiFi. 

Any Ideas?

kk-junk, 2022-06-21:
I have the same problem on my C740 with recovery image r101, rollback or use recovery image v100 until there is fix.

rama982, 2022-06-26:
Same issue with my sandy bridge laptop.

RedcatPhoenix2, 2022-07-04:
On C740, I recommend using Rammus r94. As Rammus 95+ has some issues. It may work via updating to r101 (Rammus) using chromeOS-Install or the Brunch PWA

rayep, 2022-08-28:
Same issue as https://github.com/sebanc/brunch/issues/1549