sebanc/brunch #1571: Chrome OS repeatedly restarting.

Anuragisme, 2022-06-28:
Chrome OS is restarting again and again after the brunch logo. How can I solve this? I am sharing the video with you.

Video Link : https://photos.app.goo.gl/Yn6jTmGtbPTjB2Pt6

Anuragisme, 2022-06-30:
@sebanc Could you help me ?

Anuragisme, 2023-01-17:
@tejasraman Please help me. This issue started when I turned on PWA. Whenver I turn on PWA this issue arrives.

tejasraman, 2023-01-17:
Hi @Anuragisme I want some basic info. What - is your
- model of CPU
- brunch version
- recovery image

Does this occur if the pwa setting is off?

- Tejas

Anuragisme, 2023-01-17:
CPU --- AMD Athlon Silver 3050U
Brunch Version --- 106
Recovery Image --- Zork

No, this did not occur when pwa setting is off.

Anuragisme, 2023-01-17:
My system works normally until I keep the pwa setting off but when I turn it on, the system gets stuck on the Chrome OS logo and keeps on restarting form that step.

tejasraman, 2023-01-17:
Just as a troubleshooting measure update brunch to 107 without the PWA and see if the PWA option works then.

To do this, download brunch from the github, open crosh (ctrl+Alt+t), and type these 2 commands in order:
```
shell
cd ~/Downloads
sudo chromeos-update -f yourchromeosfile.tar.gz
```
(substitute the name of the downloaded file for that file)

Anuragisme, 2023-01-19:
No, the problem is not solved it again got stuck at chrome Os logo which
slowly turn fainter to bolder.

Anuragisme, 2023-01-19:
@tejasraman

Anuragisme, 2023-01-19:
Thank you so much now the issue is resolved. I reinstalled the Chrome OS with latest version files and when I turn the PWA setting on it was working well. 

Thank You so much @tejasraman

tejasraman, 2023-01-20:
You're welcome!

closed 2023-01-20 by Anuragisme