sebanc/brunch #1573: Support touch screen for surface pro 7+, 8

nhatminhhp1999, 2022-07-01:
surface linux team has developed new driver for surface pro 8 touch screen.
 Can you add it to your project. 
"https://github.com/quo/ithc-linux"

sebanc, 2022-07-15:
I tried to include it in the below build, but I have no way to test it:
https://github.com/sebanc/brunch-unstable/releases/tag/r103-unstable-20220715

 Enable "ithc_touchscreen" option.

nhatminhhp1999, 2022-08-01:
> Tôi đã cố gắng đưa nó vào bản dựng bên dưới, nhưng không có cách nào để kiểm tra nó: https://github.com/sebanc/brunch-unstable/releases/tag/r103-unstable-20220715
> 
> Bật tùy chọn "ithc_touchscreen".

It work! thank

nhatminhhp1999, 2022-08-01:
> Tôi đã cố gắng đưa nó vào bản dựng bên dưới, nhưng không có cách nào để kiểm tra nó: https://github.com/sebanc/brunch-unstable/releases/tag/r103-unstable-20220715
> 
> Bật tùy chọn "ithc_touchscreen".

i can't remote " shelf" in chromeos 
it does not respond  It's like I didn't press any buttons

Robbysseus, 2022-08-12:
I have options="ipts_touchscreen" & kernel "/kernel-5.10"

but the touchscreen/stencil is not working, and neither is my right mouse click (touchpad).
I have Surface 7 pro with 10th gen i7

could someone help me out?

nhatminhhp1999, 2022-08-12:
> I have options="ipts_touchscreen" & kernel "/kernel-5.10"
> 
> but the touchscreen/stencil is not working I have Surface 7 pro with 10th gen i7
> 
> could someone help me out?

disable ipts and enable ithc options i think can run

colche, 2023-02-03:
> i can't remote " shelf" with touchscreen in chromeos it does not respond It's like I didn't press any buttons

Thankyou the options="ithc_touchscreen" worked to get the touch working on surface pro 7+. (Running r109 stable 20230201 and v109 "volteer")
I also have "force_touchscreen" option which shows  apps in the traditional tablet app drawer, but reverts to the bottom left desktop menu when it detects input devices plugged in.

I also can't touch any of the icons on the "shelf" - (bottom task/toolbar) though. It just ignores the touch. They are clickable with a plugged in mouse. I've tried removing the "force_touchscreen" option and a powerwash but the shelf is still not interactive in desktop mode.

Anybody had any luck with getting touch interaction on the shelf with the ithc_touchscreen driver?

nhatminhhp1999, 2023-02-03:
> > i can't remote " shelf" with touchscreen in chromeos it does not respond It's like I didn't press any buttons
> 
> Thankyou the options="ithc_touchscreen" worked to get the touch working on surface pro 7+. (Running r109 stable 20230201 and v109 "volteer") I also have "force_touchscreen" option which shows apps in the traditional tablet app drawer, but reverts to the bottom left desktop menu when it detects input devices plugged in.
> 
> I also can't touch any of the icons on the "shelf" - (bottom task/toolbar) though. It just ignores the touch. They are clickable with a plugged in mouse. I've tried removing the "force_touchscreen" option and a powerwash but the shelf is still not touch interactive in desktop mode.
> 
> Anybody had any luck with getting touch interaction on the shelf with the ithc_touchscreen driver?

after many time, I found out this is an error coming from the ITHC driver. Even linux get this error

colche, 2023-02-03:
> after many time, I found out this is an error coming from the ITHC driver. Even linux get this error

what error do you mean? The touch seems to be working in linux for me (Ubuntu 20.10 with linux-surface package)
which suggested to me it was more an oddity with the chrome os shelf and how it is trying to read touches...?

nhatminhhp1999, 2023-02-03:
X11 windows system not working or sometime misunderstanding touch. I think "shelf" use X11 not wayland, you can try ubuntu with X11. If everything fixed, we need update ITHC driver new version.
Hope everything fixed.
Feedback in here for developer: https://github.com/quo/ithc-linux/issues/5

colche, 2023-07-26:
Can report the touch on the shelf is working for me on brunch r113 - have not tried r114 yet (Surface Pro 7+)

Cyanoxygen, 2023-08-09:
Hi,
I have a Surface Pro 7+ which uses ITHC, the option `ithc_touchscreen` worked but without multitouch support, only one finger is detected, and no pinch-to-zoom gesture can be performed. Does Brunch kernel support multi-touch on ITHC currently?

I am running the brunch 6.1 kernel. Any suggestion could help, thanks!