sebanc/brunch #1575: Boot menu

mhdrasel, 2022-07-02:
Is it possible to get the "Google Chrome OS" name in the boot options menu like in the photo 
[image]

GerardoHD, 2022-07-04:
Hi! Yes, it's possible.

You just have to:

Press ctrl-alt-t

Execute these three commands:

shell

sudo su

efibootmgr --create --disk /dev/sda --part 1 --label "ChromeOS" --loader "\efi\boot\bootx64.efi" --verbose

You might have to change the partition number, but you can get that by executing lsblk -e7.

Hope it helps

dracinn, 2022-08-12:
I have a hackintosh with Opencore will this command get rid of No Name and is --verbose needed or could I exclude it?

mhdrasel, 2022-08-15:
@GerardoHD thank you for the command but its not work on my device ...

GerardoHD, 2022-08-16:
Hi, i'm sorry but i don't have any experience with Opencore or Hackintosh, so i can't help any further. Hope someone else can chime in and help you get this issue solved soon.
I recommend that you share more information about your device other than saying that it's not working so someone with more experience might feel inclined to try to help. Good luck.