sebanc/brunch #1578: Hardware acceleration disabled.

blintwave, 2022-07-04:
Since few releases, thinking it was around Brunch version 98/99 or so, I could not get hardware acceleration. I always used a Rammus image with my laptop and always worked before.

My laptop has a n3060 cpu with intel hd 400 gpu build in ( I believe). Just to say using Brunch in the past it always worked. And recently I installed arch linux with libva-intel and hw acceleration worked. I even installed Chrome OS Flex and also got hw acceleration.

I wonder if more people has a problem with it. Personally I tested playing Stadia, you can really notice the difference btw. And also checked on youtube and also shows hw not working anymore.

Also for people notice a drop in battery life time this could be the problem. Watching alot of youtube without HW makes your cpu run higher and gives you less battery.....just saying.

tejasraman, 2022-07-12:
n3060..... this is strange. My device that I just installed brunch on has an N3060 had brunch + HW accel. 
Try an octopus 101 image (that one had the best support for my case) with brunch 102 and try the nomodeset cmdline parameter (let's see) in brunch config.

And did you install arch yourself?

blintwave, 2022-07-14:
> n3060..... this is strange. My device that I just installed brunch on has an N3060 had brunch + HW accel. 
> Try an octopus 101 image (that one had the best support for my case) with brunch 102 and try the nomodeset cmdline parameter (let's see) in brunch config.
> 
> And did you install arch yourself? It's not the easiest thing in the world but is pretty good on a functional basis

I always used rammus and never had problems with HW acceleration in the past. I could try your solution but dont have the time yet.

Btw installing arch is pretty straight forward using archinstall command, got up and running system less then 20 minutes. Or for people thinking it is still to difficult they can try manjaro or something alike.

tejasraman, 2022-07-14:
I did it recently. It was a pain because I had to put Ethernet drivers on a usb and use them manually.

If to still have brunch installed, open brunch config and add nomodeset as a boot parameter

blintwave, 2022-07-16:
I tryed the nomodeset but during startup ( brunch loading screen ) system hangs and does not go further. Perhaps I am doing something wrong. I used the octopus 101 like you said.

tejasraman, 2022-07-16:
Enable verbose boot and take a video of the log please.

blintwave, 2022-07-16:
I used verbose a second time but no errors comes up... Gonna try a different image next time once i have some spare time. If it hangs again i can see if verbose outputs a different result.

P.s. which kernel did you use btw?

tejasraman, 2022-07-16:
All of them worked, 5.10 worked best.
Can you take a verbose video before changing anything and without nomodeset?

blintwave, 2022-07-16:
Good idea...Not at home right now but I will post the result once I have time and I will try the rammus image too, just to see if there might be a difference. 👍

blintwave, 2022-08-19:
Just an update.

I got HW acceleration working again. After a tip on this thread https://github.com/sebanc/brunch/issues/1548 I installed a coral recovery image which supported my cpu/gpu ( see first post for specs ) in a better way in combo with the latest Brunch 103 version. I just installed it the normal way without changing anything to the brunch config afterwards.

Tho it still makes me wonder why the Rammus image is not working anymore? In the past I always had HW acceleration with it but not anymore. Could it be that the OEM's are modefying the recovery images for hardware support or does Brunch itself do something with hardware support ( depending on kernel )

tejasraman, 2022-08-19:
Oh…. I remember why. Some OEMs are replacing i915 with iHD in the recovery images.Thanks @Brendon!

blintwave, 2022-08-19:
> Oh…. I remember why. Some OEMs are replacing i915 with iHD in the recovery images.Thanks @Brendon! 

So...is there a way to switch between the 2 drivers if you are on rammus for example? If not it means that allot of hardware will be obsolete if oem's change to iHD.

As suggestion tho maybe within brunch config one could select the right driver for your gpu perhaps.

tejasraman, 2022-08-19:
Remember brunch is not the intended way to run ChromeOS.
ChromeOS flex still has (and will have) i915 for the foreseeable future

blintwave, 2022-08-20:
> Remember brunch is not the intended way to run ChromeOS.
> ChromeOS flex still has (and will have) i915 for the foreseeable future
> 

I know Brunch is not the official way, its the reason why I use it.

But since Brunch is running on Linux kernel can gpu driver support not be added like say...wifi drivers or is it technical not possible that way?

Just asking tho, I am not tech savvy with kernel development but I find it interesting to know how it works a bit.

tejasraman, 2022-08-20:
Not really... there is no i915 support in ChromeOS

brendon, 2022-08-21:
Wrong Brendon :)

tejasraman, 2022-08-22:
That's how it shows in my email client...
And outlook messed up formatting.
Sorry @brendon