sebanc/brunch #1580: Screen auto rotation not working with 5.15 kernel Surface 5

Steso871, 2022-07-08:
Hello!

I have a problem with screen auto rotation with 5.15 kernel on my Surface 5 (2017)

With any other kernel its working perfect but not with 5.15.

Im on 102.0.5005.125 os and rammus R102 but i had this issue with 90 chrome os aswell.

s256, 2022-08-31:
Same hardware, same versions, same issue here. Manual screen rotation works. 
I'm using linux dual boot setup.
Second OS: 
Fedora with surface kernel, which runs kernel. 5.18.18-1 right now and with this kernel screen auto rotation works.

yasurok, 2022-12-03:
I too have installed chromeos_15117.112.0_rammus on my Surface Pro LTE Advanced (2017, 5th gen), and regret to report I am experiencing the same problem.

2wenty2wo, 2023-02-12:
Hi,

Same hardware and issue here.