sebanc/brunch #1581: goodixgt9110P not working on iota flo360

gingerjedi22, 2022-07-09:
Hi, using 101 rammus, with kernel 5.4, have enabled goodix, and tried enabling touch with the search+T command, to no avail.

I have managed to get bluetooth and wifi working.

Can anyone guide me as to what I am doing wrong.

Screens working fine on W11.

Kind Regards.

gingerjedi22, 2022-07-09:
i have enable goodix on the 5.4, 5.10. and 5.15, and no joy.

tejasraman, 2022-07-12:
Try `modprobe goodix` in a shell (go to crosh and type "shell" and inside shell run that command) - to get to crosh, Ctrl+Alt+T

gingerjedi22, 2022-07-14:
Thanks tejasraman, my skill level here is very weak but i will have a go, your help is so much appreciated, kind regards.

tejasraman, 2022-07-14:
That is to see if the goodix kernel module (basically, a driver) is being used. Oh, I made a mistake. Run this instead; `sudo modprobe goodix`

tejasraman, 2022-07-14:
I'll send a video if needed

gingerjedi22, 2022-07-14:
going to attempt a fix tonight, im sure when i compiled I had entered the right commands, I got the wifi and bluetooth modules working, and selected the goodix option, hopefully i can find the right commands to enable it or recompile another build, this is for a midi touchscreen sampler app called remixlive, i have 3 control surfaces for midi, and it would be useful to employ the touchscreen effects section the app has

gingerjedi22, 2022-07-14:
so, i open cosh, type in shell then the sudo command line ?

tejasraman, 2022-07-14:
Yes, and type `sudo modprobe goodix` there and see if it works. If so, we can create a boot override so that it works without that command

gingerjedi22, 2022-07-14:
mmm, nothing

tejasraman, 2022-07-14:
Run it and take a pic please

gingerjedi22, 2022-07-14:
no, sorry, no response

tejasraman, 2022-07-14:
Try `lsmod | grep goodix`
Also, try disabling goodix in brunch config and try modprobe again

tejasraman, 2022-07-14:
Tell me the output of the lsmod command first.....

gingerjedi22, 2022-07-14:
command not found, not sure iv got the command right

tejasraman, 2022-07-14:
It is L S M O D | G R E P  G O O D I X
`lsmod | grep goodix`

gingerjedi22, 2022-07-14:
[image]

gingerjedi22, 2022-07-14:
lo, i got there lol

tejasraman, 2022-07-14:
Weird... the goodix driver is loaded on your system.
Try disabling enable_goodix in the brunch config and then running the `modprobe goodix` command again

gingerjedi22, 2022-07-14:
trying without goodix enabled, 5.1.

tejasraman, 2022-07-14:
Do 5.15 and no goodix
And then do the modprobe to load goodix manually

gingerjedi22, 2022-07-14:
k

tejasraman, 2022-07-14:
Thanks. Report back after running modprobe

gingerjedi22, 2022-07-14:
5.1 modprobe not working
trying 5.15

tejasraman, 2022-07-14:
Remember, modprobe will be blank. Please try the screen after modprobe

tejasraman, 2022-07-14:
Kernel 5.1? What version of brunch are you on?????

gingerjedi22, 2022-07-14:
it was blank, but no touchscreen response
and my wifi failed to load, trying again

gingerjedi22, 2022-07-14:
5.15 was a no go, trying 5.4

tejasraman, 2022-07-14:
Try to modprobe after every one please

gingerjedi22, 2022-07-14:
going back to 5.10, the wfi, bluetooth, and andriod launcher all work, will remove goodix and try modprobe

gingerjedi22, 2022-07-14:
[image]
5.10 goodix not selected

gingerjedi22, 2022-07-14:
no touch

gingerjedi22, 2022-07-14:
im gonna check my windows install, just in case there has been a bios change

tejasraman, 2022-07-14:
Ok..... try chrome os flex and report back pls.

tejasraman, 2022-08-09:
Chrome os flex?

gingerjedi22, 2022-08-10:
no mate i will purchase a tablet or chromebook, that easily solves my problem

tejasraman, 2022-08-10:
Ok