sebanc/brunch #1584: Can I update brunch to r102 with ChromeOS Samus r91?

SpecterGames, 2022-07-13:
As shown on the title, will it brick my ChromeOS installation or will it work?

robertoandrade, 2022-07-27:
also interested to learn more about people's attempts at getting Samus updated to anything past Chrome OS r91. Trying to get at least kernel 4.19 loaded on it.

robertoandrade, 2022-07-27:
Been seeing some positive feedback on reddit: https://www.reddit.com/r/chromeos/comments/sv5dvt/comment/i2jmb0e/?utm_source=reddit&utm_medium=web2x&context=3

SpecterGames, 2022-07-30:
@robertoandrade Thanks for the reply, I actually tried it on my Intel i5-3450 processor, and when I updated the brunch to the latest and after a reboot, it gets stuck at the "Loading Brunch Framework..." screen for only ONE second, then my pc automatically reboots again.

robertoandrade, 2022-07-30:
Ok gonna try it on my i7 to see if I have better success and will report accordingly.

tejasraman, 2022-08-09:
IvyBridge? Start using rammus - if it works on my sandybridge laptop, it'll work on IvyBridge.